cyoung/stratux #676: Change EFIS out to GPIO?

MolsonB, 2017-10-26:
Is there a way to change the EFIS out to use the GPIO serial?  /dev/ttyAMA0
I have a TTL-Serial converter that I used SerialOut branch last year to get working. Notice the branch got merged into the main file now.

I looked in network.go @ line 243 and thought I could change it there.
`	serialDev := "/dev/serialout0" //FIXME: This is temporary. Only one serial output device for now.
`

The Makefile compiles with errors and I read it's not in sync anymore with the latest build. Not sure how else to compile the program ?

> Makefile:17: recipe for target 'xgen_gdl90' failed
> make[1]: *** [xgen_gdl90] Error 2
> make[1]: Leaving directory '/root/stratux'
> Makefile:14: recipe for target 'all' failed

MolsonB, 2019-03-14:
This is still working with latest 1.5b2 release.  Also discovered GPS still works too (didn't use it before, but now I am).

cyoung, 2019-03-14:
How reliable has it been for you?

MolsonB, 2019-03-15:
So far so good. I'm only ground testing right now, hope to be flying in the next month. But over a year working so far. Helps with the GRT EFIS, as I can use a USB instead of a serial port. I was under the assumption that I couldn't use GPS with serial out, but it is working fine. Maybe because I hardcoded the serialDev deivce in there.

Using an Amazon USB to TTL Serial Cable. The only thing if you buy the AHRS chip, it sits overs the TTL pins, so had to make mods for that.