cyoung/stratux #677: Stratux 1090ES & UAT drivers

assasen11, 2017-10-30:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ ] yes
      - [x ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

I just ended up buying the Stratux 1090ES & UAT themselves. I was hoping to use them for WIFI purposes. Where would i be able to get the drivers to manually install them?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2017-11-08:
These are software defined radios. They are receivers that can be tuned from ~100 MHz to 1.5 ghz. 

They are not associated with wifi in any way. If you want to use them to listen to radio signals on a desktop computer you should look up SDR#.

cyoung, 2017-11-13:
@assasen11 - might have the wrong product.

closed 2017-11-13 by cyoung