cyoung/stratux #679: Refuses to build

tonyhain, 2017-11-13:
1. Stratux version: git clone https://github.com/cyoung/stratux  Nov 12 2017 ~2300 UTC
Rpi 3    Linux adsb 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux

2. Stratux config: can't get that far

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ x] yes  VK-162  DIYmall
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source:  2.5a 5v USB  ::  Vilros dual kit on Amazon

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)   Avare 7.8.5

    EFB platform: (e.g., iOS 9.2)      Android 5.1.1

    EFB hardware: (e.g., iPad Mini 2)     Galaxy Tab 4

4. Description of your issue:

SD card distributed with Vilros kit on Amazon is partitioned at <2GB, with stripped down Raspbian and invalid repos. Stock Raspbian on 16GB SD works fine, but it is impossible to build stratux due to undocumented dependencies, incomplete and inconsistent recipes here and on reddit. then the final straw, static linking without indicating the library version. After many attempts and stumbles over missing dependencies cd stratux && make returns:
...
cd dump1090 && make
make[2]: Entering directory '/usr/local/src/stratux/dump1090'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/usr/local/src/stratux/dump1090'
go get -t -d -v ./main  ./godump978 ./uatparse ./sensors
go build -ldflags "-X main.stratuxVersion=`git describe --tags --abbrev=0` -X main.stratuxBuild=`git log -n 1 --pretty=%H`" -p 4 main/gen_gdl90.go main/traffic.go main/gps.go main/network.go main/managementinterface.go main/sdr.go main/ping.go main/uibroadcast.go main/monotonic.go main/datalog.go main/equations.go main/sensors.go main/cputemp.go
go get -t -d -v ./main
go build -ldflags "-extldflags -static -X main.stratuxVersion=`git describe --tags --abbrev=0` -X main.stratuxBuild=`git log -n 1 --pretty=%H`" -p 4 main/fancontrol.go main/equations.go main/cputemp.go
# command-line-arguments
/usr/local/go/pkg/tool/linux_arm/link: running gcc failed: exit status 1
/usr/bin/ld: cannot find -lwiringPi
/tmp/go-link-781636819/000002.o: In function `_cgo_709c8d94a9f9_C2func_getaddrinfo':
/workdir/go/src/net/cgo_unix.go:54: warning: Using 'getaddrinfo' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
collect2: error: ld returned 1 exit status

Makefile:21: recipe for target 'fancontrol' failed
make[1]: *** [fancontrol] Error 2
make[1]: Leaving directory '/usr/local/src/stratux'
Makefile:14: recipe for target 'all' failed
make: *** [all] Error 2
root@adsb:/usr/local/src/stratux#  warning: Using 'getaddrinfo' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
bash: warning:: command not found

to get to this point I had installed Bison to build NTPsec, then apt-get install gpsd ---
     remote           refid      st t when poll reach   delay   offset   jitter
*172.24.144.13   .PPS.            1 u   70  128  377   1.2655 -53.8831   6.7673
+SHM(0)          .GPS.            0 l   29   64  377   0.0000 -111.479   2.2816
 SHM(1)          .PPS.            0 l    -   64    0   0.0000   0.0000   0.0019
ntpd ntpsec-1.0.1+52 2017-11-09T20:58:31Z     Updated: 2017-11-12T22:45:50 (32)
the pps line is not connected yet. It is not clear to me that the stratux use of the gps is consistent with gpsd, though I did see a /dev/SHM device, so who knows.

following this wiki i simply did ---
git clone https://github.com/cyoung/stratux  && make

when that failed I backed up and added ---
git clone https://github.com/steve-m/librtlsdr

when that failed ---
git clone https://github.com/tgogos/rpi_golang.git  # following bootstrap to build 1.5, then found
wget http://dave.cheney.net/paste/go1.5.3.linux-arm.tar.gz
GOROOT=/usr/local/go
export GOROOT
GOPATH=$GOROOT/bin
export GOPATH
go version go1.5.3 linux/arm

cd /usr/local/src/librtlsdr
mkdir build && cd build
cmake ../
make
make install

along the way this list got installed, but I have no idea what order ---
apt-get install -y git subversion mercurial bzr cmake libusb-dev libusb-1.0 libtool 

the fan control / ahrs module is installed
the case is a smarticase  with R-pi 7" touch screen installed & a back to hold the fan over the heat sinks
https://smile.amazon.com/gp/product/B01HKWAJ6K/ref=oh_aui_detailpage_o00_s04?ie=UTF8&psc=1
at the moment the SDR modules are on an external USB hub for power and heat distribution, and to get the antennas next to the window. With the Vilros SD card I was able to connect Avare to the SSID, and get a gps fix, but the Avare-external app only showed intermittent readings on one SDR. When system updates were failing due to invalid repos, coupled with the lack of space I gave up on that card. I have it mounted as a usb disk on /mnt, but can't see any obvious way to find the components installed there to copy to the stock Raspbian system. The 18 month old git clone is in /mnt/root/stratux, so I could try to build that and see if it installs, but i would rather focus on current.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2017-11-13:
I don't know what Vilros is putting on their SD cards now. You can just re-image the SD card with the most recent release from the [releases page](https://github.com/cyoung/stratux/releases).

The release images should have the static libraries and most other packages needed. It doesn't have the Go compiler, you can get it [here](https://golang.org/dl/). Untar into `/root/go`. Make sure to get the "linux-armv6l" version.

I didn't read the whole issue, but it looks like from the start that you are trying to compile without gcc installed on the system...

tonyhain, 2017-11-14:
Thanks for the quick reply. I figured it was probably something simple, but since I don't know was is in stock Raspberian, and the wiki page here says git clone & compile, I started out on the wrong path and it only got worse. I don't have time to deal with it tonight, but I will try a recent release on another card. gcc is in /usr/bin, but there may not be appropriate variables defined to go with it.

tonyhain, 2017-11-21:
Must have been a bad image on the original card. installed 1.4r3 and it expanded correctly and came up.

closed 2017-11-21 by tonyhain