cyoung/stratux #680: Non-compliant GDL90 Ownship report behavior

cyoung, 2017-11-15:
Ownship report not sent when GPS position fix not valid.

https://github.com/cyoung/stratux/blob/master/main/gen_gdl90.go#L287-L289

GDL90 "3.4" says:

```
The Ownship Report is output by the GDL 90 regardless of whether a valid GPS position fix is
available. If the ownship GPS position fix is invalid, the Latitude, Longitude, and NIC fields in
the Ownship Report all have the ZERO value. 
```

cyoung, 2018-01-17:
Ref #348.

GitUser275, 2018-02-11:
Any estimate as when this will be addressed?  I'm still hoping to be able to use my Stratux without a GPS.     I don't need AHRS and I have a perfectly good GPS in the display device.  What I do need is transmission of ownship pressure altitude.

Thanks

UKJohn, 2018-04-17:
We now have EFB's that allow us to use Stratux in the UK -   they now support GDL90. So thats good.
However there is a problem with ownship pressure altitude reporting to a "free" EFB called EasyVFR - you can download it from the Appstore for your ithingy.

So the EFB  reports a band of altitudes of a threat aircraft. They say they do this because no valid pressure altitude is received from Stratux and so they display a band of altitudes covering a range of 2200'  (i.e low QNH - high QNH).

Is this the same problem reported here - or should I start a new thread ?