cyoung/stratux #682: Dangerzone Builds Not Showing AHRS on Foreflight

pp4life, 2017-11-21:
1. Stratux version:  Tried update-stratux-v1.4r3-a0d9a4be25.sh and  update-stratux-v1.4r3-3b57564490.sh

2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ X] yes
      - [ ] no
      type: GPYes

    AHRS
      - [X ] yes
      - [ ] no

    power source: Anker 

    usb cable: StarTech Cable

3. EFB app and version: Foreflight 9.4.3

    EFB platform: (iOS 11.0.3)

    EFB hardware: (iPad Mini 2)

4. Description of your issue:
I tried the two latest versions of the DangerZone builds as noted above and can't get any AHRS information. GPS works. See attached images. Seems curious that it says "Genuine Stratus: No". I've had AHRS working before in Foreflight with DangerZone builds, albeit kind of jerky. I don't remember it showing 2 devices in the Devices area before (Stratus 2 and ADS-B). Maybe I'm not remembering that correctly though. Let me know if there's anything else I should try.

[image]
[image]
[image]
[image]

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

zanecane53, 2017-11-27:
I’m having the same issue here 

Version v1.4r3 (a0d9a4be25)
ForeFlight v9.4.3
iOS 11.0.3

pp4life, 2017-11-27:
I turned my Stratux on about an hour ago and left it sitting on my desk. I just now connected my iPad to the Stratux WiFi and I get AHRS now in Foreflight. Still says it's not a genuine Stratus.

cyoung, 2017-11-27:
It has been saying that it's not a genuine Stratus for a while, seemed to not have any effect before. Will give this a try.

pp4life, 2017-11-27:
I have no idea why it wouldn't work before. Maybe I didn't have a GPS fix for long enough? Seems to be working now. The heading indicator keeps going in and out, but I'm also not moving. I assume it needs to be moving to get a decent heading since it's using GPS coordinates to determine heading, correct?

pp4life, 2017-11-28:
Confirmed, heading indicator needs motion to work properly in Foreflight. I get an X through through heading as soon as I stop moving in the car. Starts showing heading again as soon as I start moving.

rex199, 2018-01-09:
My device did not show up gps heading

cyoung, 2018-01-17:
Seems to be working ok with v1.4r3-a0d9a4be25, FF 9.5.3, iOS 11.2.2.

Make sure to upgrade everything. Check AHRS status at http://10.26.36.1/#/gps. Make sure you have a GPS lock. Heading data won't be valid/showing unless moving.

closed 2018-01-17 by cyoung

Nayajja, 2018-01-23:
I have a Stratux with Foreflight, same  versions, same story. Versions v1.4r3-a0d9a4be25, FF 9.5.3, iOS 11.2.2.

It used to work fine, then my SD card failed. I got a new SD card, and re-loaded eveything, with the new versions. GPS is locked and valid, everything works on 192.168.10.1, but ForeFlight shows no attitude info.