cyoung/stratux #684: No Tower Reception

kjohnson2289, 2017-11-22:
1. Stratux version:  v1.4r3 (bd10be4f63)
2. Stratux config:  

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [ ] yes
      - [X] no
      type:

    AHRS
      - [ ] yes
      - [X] no

    power source:  Anker PowerCore 10000

    usb cable:  Anker

3. EFB app and version: Foreflight 9.2.1

    EFB platform: iOS 11.1.2

    EFB hardware: iPad 3

4. Description of your issue:

Barely/Rarely pickup ADS-B Towers.  Flying an MV-22 Osprey up to 9,000' AGL.  Stratux placed in cockpit at upper windshield.  UAT shows 0 messages.  1090ES shows plenty of messages.  Can see some traffic.  Located in Jacksonville, NC flying in Eastern NC

Nokomis449, 2017-11-23:
First the easy stuff.  Log onto the Stratux wifi and open a browser (Safari, Chrome, etc.).  Navigate to 192.168.10.1 and find the "Settings" tab.  Verify that the "978 Mhz" option is ON.  If it's not, turn it on and hopefully you've found the problem.  If it was already ON, then go eat at Bojangles across from Lowes and take the Stratux with you.  While sitting there, you'll be able to pick up the ADS-B tower that's a few hundred meters to the NNE of you on the other side of the pond.  If you can pick up that tower, then the Stratux is working as expected.  If it does NOT pick up that tower, then let's diagnose further.  Turn the Stratux off and remove the 978 USB radio (the "dongle" - I HATE that word!) from the Stratux and turn the Stratux back on.  In the absence of a 978 radio, the 1090 radio will now revert to the 978 frequency.  If you now begin to receive the tower, you know that the 978 radio is bad and (hopefully) can be returned/exchanged.  If these ground tests are all successful and indicate that the radio's are both working properly, then the issue could be something like heated windows in the Osprey blocking the signals - this happens on some civilian jets; don't know what the Ospreys use.  If possible, try moving the Stratux to another window that may have a better view/not heated.  FWIW, the Stratux has been used successfully in a C17 and a B1, it'd be great to add Osprey to that list.

kjohnson2289, 2017-11-26:
1.  the 978 MHz antenna/SDR is on.
2.  Took a trip to Lowes today (across from Bojangles and started picking up the tower about 10 or 15 miles (driving) to the north.  I had constant communication with the antenna thereafter (until I left Lowes and traveled back 10 or 15 miles to the north).  Could anything else be wrong with my stratux that  would allow me to receive signal within 10 miles on the ground but not in the air?
3.  We do have heated windshields.  Is the construction of the windshield typically the problem or the actual application or power/heat to the windshield?  I'll play with it a little bit and try another window (though I have in the past).
[image]
[image]
[image]
[image]
[Log File.pdf](https://github.com/cyoung/stratux/files/1504284/Log.File.pdf)

rhole, 2017-11-26:
Have you considered ignition noise?  Distance from your antenna to transponder antenna?

Rick & Sonia Hole
Email: r.hole@ieee.org<mailto:r.hole@ieee.org>

[img_6157]<https://user-images.githubusercontent.com/33913567/33245123-6a92d298-d2d1-11e7-9a6e-209c6b46b755.PNG>
[img_6159]<https://user-images.githubusercontent.com/33913567/33245124-6ab2479a-d2d1-11e7-96bf-297b81eb4a12.jpg>
[img_6163]<https://user-images.githubusercontent.com/33913567/33245125-6ac90a20-d2d1-11e7-928b-84bc92a5a7b6.jpg>
[img_6169]<https://user-images.githubusercontent.com/33913567/33245126-6ade18de-d2d1-11e7-81f1-f52b48702968.PNG>

cyoung, 2017-12-01:
@kjohnson2289 - what does your build look like? Any possibility that the antenna wires got twisted and one of them is intermittent? Maybe try switching those around.

kjohnson2289, 2017-12-05:
Well, I went flying last week and was at 7 to 8k feet with great success. I had a maximum of 5 towers and a minimum of 1 to 2 towers for most of the flight. I played with antenna placement quite a bit so I’m wondering if the heated windscreens was, in fact, the issue from the start.  Would high gain antennas help at all?  I noticed that the jets discussed on the main page were tested using high gain antennas.

cyoung, 2017-12-05:
That should help... what are you using now?

cyoung, 2018-01-17:
You should have high gain antennas if you got the kit above. Sounds like you got it working. Re-open if any other issue.

closed 2018-01-17 by cyoung