cyoung/stratux #685: 192.168.10.1/traffic

cbfraser, 2017-11-25:
1. Stratux version:v1-4r1 (f6bad36cdc)
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
192.168.10.1/traffic as described in 
https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.md
 returns  "not websocket protocol", /getTraffic & variants returns page not found
all the other 192.168.10.1/get*  pages work
this would be helpful in testing audio alerts

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

GitUser275, 2017-11-29:
I asked essentially the same question and it was answered here:

https://www.reddit.com/r/stratux/comments/7f0912/help_using_ws192168101traffic/

cyoung, 2018-01-17:
@cbfraser - note "ws" in `ws://192.168.10.1/traffic` -- you need to connect with the WebSocket protocol, not HTTP.

closed 2018-01-17 by cyoung