cyoung/stratux #687: WiFi not available

dmireles24, 2017-12-13:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ x] dual

    GPS
      - [ x] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ x] no

    power source:
Power Bank
    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
Fore Flight
    EFB platform: (e.g., iOS 9.2)
iOS 11
    EFB hardware: (e.g., iPad Mini 2)
iPad New
4. Description of your issue:
Yesterday while flying everything was fine. I was connected and receiving weather and traffic. The red LED was lit and the green LED was flashing indicating a WiFi connection. In the second hour of flight the Stratux fell off WiFi. The green LED was steady on and I am now unable to see it in the list of available WiFi devices. Is this fixable or a hardware issue that needs to be replaced?

pp4life, 2017-12-13:
I believe a flashing green light IS BAD, at least in the more recent versions of the Stratux software. There is also a flashing green light on most GPSs when it has a fix. Make sure that's not what you're seeing flashing.

dmireles24, 2017-12-14:
Current status is solid green and red. there is no flashing. When powered up both lights are steady on. When I try to connect via WiFi the stratux is no longer listed. It was when I started the flight then it dropped off and I have been unable to connect again with any device. I have not updated any software on the Stratux. Thanks so much for your help.

emuyshondt, 2017-12-14:
When something similar happened to me, it turned out the SD card was
corrupted and broken. I could read the directory, but I could not reflash
it. I had to buy a new SD card. Once I installed that, my Stratux worked
normally again.

dmireles24, 2017-12-26:
ill try the sd card idea. I just dont want to buy a new Raspberry and it be the card.

dmireles24, 2017-12-26:
Okay I just reflashed the SD card. All is working well. Thanks for the help.

closed 2017-12-26 by dmireles24