cyoung/stratux #688: Dirty bit is always found set at system boot

GitUser275, 2017-12-17:
1. Stratux version: 1.4r3
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [x ] no
      type:  NA

    AHRS
      - [x ] yes
      - [ ] no

    power source: Bench power supply feeding regulator board set to 5.25 volts

    usb cable:  short and red LED is on steady

3. EFB app and version: (e.g., WingX Pro7 8.6.2)  NA

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

At boot a connected HDMI monitor shows messages:

[OK] Started File System Check on /dev/mmcblk0p1
[time] systemd-fsck[192]: 0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
[time] systemd-fsck[192]: Automatically removing dirty bit.

The messages say that the dirty bit was removed (cleared) so it should remain clear until set by some process or event.  If intentionally set it should be cleared as part of the system shutdown process.  
 However, if Stratux is re-booted from the UI, then the same message is seen when it starts. 

Re-boot does not involve any power supply switching and the RED LED remains lit continuously.  The same message is seen with or without GPS connected.

[image]

Axtel4, 2017-12-17:
It looks like a Raspbian issue.  There is a lot of traffic about the error and work arounds on the Raspberry Pi forums.

gituser276, 2017-12-17:
It appears the Stratux Releases have dirty boot device /dev/mmcblk0p1

Fresh load on clean SD card 16gb

root@raspberrypi:~# fsck.fat -V /dev/mmcblk0p1
fsck.fat 3.0.27 (2014-11-12)
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action
? 1
Starting check/repair pass.
Starting verification pass.
Leaving filesystem unchanged.
/dev/mmcblk0p1: 87 files, 2551/7673 clusters

root@raspberrypi:~# fsck.fat -a /dev/mmcblk0p1
fsck.fat 3.0.27 (2014-11-12)
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Performing changes.
/dev/mmcblk0p1: 87 files, 2551/7673 clusters

root@raspberrypi:~#  fsck.fat -V /dev/mmcblk0p1
fsck.fat 3.0.27 (2014-11-12)
Starting check/repair pass.
Starting verification pass.
/dev/mmcblk0p1: 87 files, 2551/7673 clusters

root@raspberrypi:~#  fsck.fat -a /dev/mmcblk0p1
fsck.fat 3.0.27 (2014-11-12)
/dev/mmcblk0p1: 87 files, 2551/7673 clusters

on reboot, the same appears, 
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.

My fresh Rpi install has no such problem. 
https://www.raspberrypi.org/blog/raspbian-stretch/

 **The issue is related to Stratux release distributions.** The boot filesystems is bad as written to a clean SD card.

kdknigga, 2017-12-17:
> The boot filesystems is bad as written to a clean SD card.

I'm not sure that makes sense.  Even if the initial image were bad, fsck
should still fix it.

gituser276, 2017-12-18:
I'm not sure that makes sense.  Even if the initial image were bad, fsck
should still fix it.

It makes sense. fsck repairs the issue only to return, check for yourself.

Try it, on reboot it is bad.

 Standard Jessica and Stretch images from Rpi no such issue

kdknigga, 2017-12-18:
My initial guess (having not actually looked at it yet) would be something
hanging on shutdown that never allows the filesystem to unmount.  When I've
seen this in the past it hung the whole shutdown process, but I wonder if
systemd is more forceful about making things stop, cleanly or not.

gituser276, 2017-12-18:
So not to guess, Regression testing starting back to when the error is taking place would be a method to find out why this is happening.

It would seem normal to migrate to the latest Rpi build, Debian Stretch. However, it maybe Stratux wishes to remain to be broken and failing when people attempt to use it. I do not know?

Axtel4, 2017-12-18:
Is this this the root cause of the above noted link? Or, is this a logged nuisance error?

gituser276, 2017-12-18:
Maybe it is Malware. No one knows the cause root or otherwise. Both 1.4r2 and 1.4r3 have this error. I ran a test this morning.

The boot Partition has errors that will not go away, not a good sign of a healthy Software operating system. Try it for yourself and let us know what happens. **Post some data Please.**

cyoung, 2017-12-18:
>It would seem normal to migrate to the latest Rpi build, Debian Stretch.

There's a [bug](https://github.com/raspberrypi/linux/issues/2134) related to RTL-SDR in Kernel 4.9. What version is Stretch at now?

GitUser275, 2017-12-18:
As the person who raised the issue I'd like to thank those who have reproduced it.  It's good to know it's not isolated to my system.   
I hope it's not confusing anyone that there are two users with very similar names commenting on this issue.  When I created my name I was not aware that a similar name was in use here.

gituser276, 2017-12-18:
Thank you GitUser275, good work.

I'm running Regression testing in a attempt to find when the error started. I'm now using v1.0r1 to conduct the evaluation

gituser276, 2017-12-18:
v1.0r1
Everything here comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

Type 'stratux-help' for a few debugging commands.
Last login: Mon Dec 18 17:02:03 2017
root@raspberrypi:~# fsck.fat -V /dev/mmcblk0p1
fsck.fat 3.0.27 (2014-11-12)
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action
? 1
Starting check/repair pass.
Starting verification pass.
Leaving filesystem unchanged.
/dev/mmcblk0p1: 172 files, 2693/7673 clusters
root@raspberrypi:~#

Axtel4, 2017-12-18:
I run my Stratux headless.  Does the require connecting to a monitor/keyboard to see the error?

gituser276, 2017-12-18:
Kernel 4.9.x drops USB data on Pi 2B
There's a bug related to RTL-SDR in Kernel 4.9. What version is Stretch at now?
4.9 is what I'm running

https://discussions.flightaware.com/t/raspbian-stretch-dump1090-mutability-v1-15-dev-does-not-work/25009
Did have issue. My Rpi3 with FlightAware works as it should now.

gituser276, 2017-12-18:
I run my Stratux headless. Does the require connecting to a monitor/keyboard to see the error?

ssh into your Rpi running on your network, wired or wireless
user sudo or the root account
enter
fsck.fat -V /dev/mmcblk0p1 to view the error

fsck.fat -a /dev/mmcblk0p1 clears the error

reboot and the error is back

gituser276, 2017-12-19:
I was not able to create boot images from stratux-v0.6r1-bbbcd3bbe1 down 

Had and old download of v0.8r2 extraction and booted
stratux-v0.8r2-db130aab76
login as: root
root@  password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Dec 19 00:37:33 2017 
root@raspberrypi:~# fsck.fat -V /dev/mmcblk0p1
fsck.fat 3.0.27 (2014-11-12)
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action
? 1
Starting check/repair pass.
Starting verification pass.
Leaving filesystem unchanged.
/dev/mmcblk0p1: 86 files, 2551/7673 clusters
root@raspberrypi:~#

Not sure what to make of it.

gituser276, 2017-12-19:
Ok folks
Mounted the SDcard in Sdcard USB port
login as: root
root@192.168.3.111's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Dec 19 00:55:55 2017 from 192.168.3.106

root@raspberrypi:~# fsck.vfat -avy /dev/sda1
fsck.fat 3.0.27 (2014-11-12)
Checking we can access the last sector of the filesystem
Boot sector contents:
System ID "mkfs.fat"
Media byte 0xf8 (hard disk)
       512 bytes per logical sector
      8192 bytes per cluster
         1 reserved sector
First FAT starts at byte 512 (sector 1)
         2 FATs, 16 bit entries
     16384 bytes per FAT (= 32 sectors)
Root directory starts at byte 33280 (sector 65)
       512 root directory entries
Data area starts at byte 49664 (sector 97)
      7673 data clusters (62857216 bytes)
63 sectors/track, 255 heads
         0 hidden sectors
    122880 sectors total
Reclaiming unconnected clusters.
/dev/sda1: 172 files, 2693/7673 clusters
root@raspberrypi:~#

as noted
**Checking we can access the last sector of the filesystem**

It appears the boot file system structure is defined in a way that the OS cannot access the last sector of the file system. Stratux defines the disk geometry for the image, The OS boot to this and the error is likely **Checking we can access the last sector of the filesystem** and sets the 
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt as it cannot determine the last sector status. Fsck clear the flag only to return on system boot.

What errors this causes as a result is unknown.

**/dev/sda2 is clean
root@raspberrypi:~# fsck -va /dev/sda2
fsck from util-linux 2.25.2
/dev/sda2: clean, 62828/150176 files, 482912/597504 blocks
root@raspberrypi:~#**

gituser276, 2017-12-19:
quite a church mouse here;

https://www.reddit.com/r/stratux/comments/7ip78c/no_adsb_wx_and_disk_bytes_used_error/

Actually I don't think the wx has worked on either of the 2 stratux I have, I'm using Ping EFB. I have troubleshooted using both Stratux and pings and different devices. Now the one that had the disk problem has no wifi. I'm not sure how to reimage the card. I bought it on Amazon as part of the kit.

It appears you folks are selling SDcards with bad disk images. Please let Amazon know this. If not, I will.

Helno, 2017-12-19:
That guy actually got in touch on slack and has a defective pingEFB. 

He was seeing 21+ towers sitting on the ground.

gituser276, 2017-12-19:
Good wishes and happy Holiday day

Do are you see fit. I will do so also.

GitUser275, 2018-01-01:
I would appreciate a Stratux developer commenting on the following:

Can you reproduce the reported behavior?

If so:
   Is the distributed image corrupted?
   Is there any impact on system operation?
   Is the setting of the dirty bit related to my frequent failure of Etcher verification when burning the Statux image?

Thanks

Helno, 2018-01-01:
I did some playing around and was able to clear the dirty bit. Only for it to come back as soon as things were rebooted. Doesn't seem to matter how it is shutdown or what hardware is present.

fsck.fat -trawl /dev/mmcblk0p1 is what I ran to clear it. Found that information here. 
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=64843&sid=6286999d44b4d36a215130dbbb7dc50c

As far as I can tell there is no corruption in a freshly written image. FSCK doesn't detect anything.

I wrote 20 SD cards before Christmas and did not see any verification failures in Etcher. They were all brand new 2 gb cards. Previously I was using 8gb cards from Microcenter and again never had a verification failure in Etcher.

cyoung, 2018-01-17:
I took a live microSD card out of a unit and ran `fsck_msdos` on the /boot partition from a Mac. Cleared the dirty bit. Booted the unit after this and the dirty bit was set yet again. Closing for now unless we note some operational issue.

closed 2018-01-17 by cyoung