cyoung/stratux #690: Stratux has not received valid ADS-B position transmissions

finnishpilot, 2017-12-22:
1. Stratux version: v1.4r3
2. Stratux config:

    SDR
      - [X ] single
      - [ ] dual

    GPS
      - [X ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X ] no

    power source: 10 000 mAh external Power Bank

    usb cable: Yes

3. EFB app and version: Avare

    EFB platform: Android

    EFB hardware: Samsung TabE 9.6

4. Description of your issue:
I have noticed during last couple of flights that I am loosing traffic from my EFB. I went to reproduce this issue yesterday by passing by car an airport is transmitting weather.
- I received METAR information
- I see traffic in EFB display
- Little later traffic disappear. 
--> Went to browser and connected to Stratux
--> Selected: Menu -> ADS-B and TIS-B traffic
--> MESSAGE: Stratux has not received valid ADS-B position transmissions from the aircraft in this section. They will not appear on your EFB map. See help page for details. 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2017-12-22:
What do your message rates look like when things drop off? It should be easy enough to reproduce sitting on the ground at home. If it works and then messages dro off to nothing it might be a SDR problem.

If you are getting messages many of them will be mode-s non position data. Anything in the upper listing will be sent to your EFB.

finnishpilot, 2017-12-23:
Hi,
When it freeze it show 0 UAT messages. And it froze after 4:04 min.- GPS hardware: Nothing- GPS solution: Nothing- GPS satellite: Nothing
Another run today:- GPS hardware: USB u-blox 7 GNSS receiver (NMEA-UBX protocol)
- GPS solution: 3D GPS, 4.6 m
- GPS satellite: 9 in solution; 12 seen; 15 tracked

I have another SDR so I can try it as well. Before I do that go back to release Stratux-v1.4r1 to see how it works.

FYI, I live close to DFW airport and we have traffic all the time.

Regards,Esa

Helno, 2017-12-23:
What about 1090 messages? The vast majority of traffic comes in over 1090.

What do you mean by Freeze? In the unit becoming non-responsive? What does the wifi do when this happens?

cyoung, 2018-01-17:
@finnishpilot - are you using another EFB on your tablet (FltPlan), etc?

Did you try updating to v1.4r4 and see how it does?

finnishpilot, 2018-01-18:
I format memory card and loaded v1.4r3 again. Other thing what I did is that I added 1090 SDR and issue has been gone since that. I have used Stratux in car and during flying and it has been working fine.

Thanks for informing that v1.4r4 is available! I'll load it before the next flight.

closed 2019-02-01 by Helno