cyoung/stratux #691: Stratux not connecting to FltPlan Go

curts63, 2017-12-30:
1. Stratux version: 1.4r3
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ x] yes
      - [ ] no
      type:

    AHRS
      - [ x] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) FltPlan Go 4.0.7

    EFB platform: (e.g., iOS 9.2) Android 7.0

    EFB hardware: (e.g., iPad Mini 2) Samsung Tab S2

4. Description of your issue:   My stratux will not connect to fltplan go on my samsung s2. The stratux app will open the device and you can see the traffic listings and gps status. When you click on stratux in fltplan, it says "connecting", but then it turns to "inactive". The stratux app still shows that its working, but won't connect to fltplan. I reflashed the micro SD card, but that didn't work. The stratux works fully on my android phone, but not the tablet.  I'm not really that savvy with technology, but thought I could handle this.
Fltplan version 4.0.7 build 3021, my stratux is version 1.4r3 tablet is Samsung Tab S2 Android version 7.0

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2017-12-31:
Have you tried getting in touch with FLTplan go? The fact that everything works on your phone means that your Stratux is operating correctly. 

Perhaps try a different EFB on your tablet to see if it is an EFB issue or an operating system issue.

curts63, 2018-01-01:
[image]

Not sure if this helps, but it looked like something is wrong?  I noticed this after turning on the logging.   I did email the support team at FltPlan, but with the holiday, I haven't heard back.  I really don't want to use a different EFB app.

Helno, 2018-01-01:
That is what happens if you turn logging on and let it run for a while. There is no need to turn logging on unless you are trying to capture a specific problem. The option is going to be moved to a hidden tab shortly as so many people turn it on for no reason. 

You can clear the saved log files by following the instructions here. 

https://www.reddit.com/r/stratux/comments/75m73d/log_filling_up_resulting_in_flashing_green/

As for troubleshooting your original issue I was not suggesting that you switch to a different EFB. What I was suggesting is that if you cannot get ADS-B data on multiple EFB's on that tablet that it is an issue with the tablet rather than the EFB.

If you can get data on your phone that implies that the issue is with the tablet. Using a different EFB allows you to figure out where the chain is broken.

curts63, 2018-01-04:
I wanted to thank you for taking the time to try and assist me.  I was lucky enough to stumble onto another site where someone had the same problem.  The answer was to turn off my “battery saver” mode on the tablet.  Believe it or not, this took care of it.  All is working good now.

 

Curt

 

https://www.reddit.com/r/stratux/comments/75m73d/log_filling_up_resulting_in_flashing_green/

Helno, 2018-01-04:
Your welcome.

Got a reference to where you saw that? It would be nice to have it to reference in case the issue comes up again.

Nokomis449, 2018-01-04:
I've seen the same solution posted within the last month or two but can't remember where.  Possibly the iFly forums but my searches are not finding it.

curts63, 2018-01-05:
I found it on the I-fly forum.  I did a google search on the issue and the forum popped up. 

https://www.iflygps.com/Support/Forum/forumid/17/threadid/30253/scope/posts/threadpage/1

Curt

cyoung, 2018-01-17:
Glad you got it fixed.

closed 2018-01-17 by cyoung