cyoung/stratux #692: Run out of SD storage space because of log file size keeping increase

rex199, 2018-01-05:
1. Stratux version: v1.4r3
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: Stratux Vk-162 Remote Mount USB GPS 

    AHRS
      - [X] yes
      - [ ] no

    power source: standard

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) forefight 9.5.2

    EFB platform: (e.g., iOS 9.2) ios 9.3.3

    EFB hardware: (e.g., iPad Mini 2) ipad mini 4 with cellular

4. Description of your issue:

Base from the log shows:  there is loop keeping input logs for this: 
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
which cause the log file sizes increase faster and the SD card (8G, after installed stratus, there are about 230-240MB free space left) will run out of storage faster, then system crash and stop works!

I tired v1.4r3 (bd10be4f63, a0d9a4be25, 3b57564490)  and v1.4r2 ae734f41ba

All of them are the same issues. Any idea what is happen and how to fix it? Thx

2018/01/05 17:21:44 Stratux v1.4r3 (a0d9a4be257d327a24ba1c3045aca52faa81836a) starting.
2018/01/05 17:21:44 Entered uatReader() ...
2018/01/05 17:21:44 read in settings.
2018/01/05 17:21:44 Developer mode set
2018/01/05 17:21:44 AHRS Info: initializing new Simple AHRS
2018/01/05 17:21:44 AHRS Info: couldn't start ahrswebListener: dial tcp [::1]:8000: getsockopt: connection refused
2018/01/05 17:21:44 AHRS Info: IMU Calibrations read from settings: accel -0.016851 1.004701 -0.215636; gyro -0.185095 1.003296 0.614795
2018/01/05 17:21:45 AHRS Info: IMU Calibrations read from settings: accel -0.016851 1.004701 -0.215636; gyro -0.185095 1.003296 0.614795
2018/01/05 17:21:46 AHRS Info: IMU Calibrations read from settings: accel -0.016851 1.004701 -0.215636; gyro -0.185095 1.003296 0.614795
2018/01/05 17:21:47 AHRS Info: IMU Calibrations read from settings: accel -0.016851 1.004701 -0.215636; gyro -0.185095 1.003296 0.614795
2018/01/05 17:21:48 AHRS Info: IMU Calibrations read from settings: accel -0.016851 1.004701 -0.215636; gyro -0.185095 1.003296 0.614795
2018/01/05 17:21:48 AHRS Info: Successfully initialized BMP280
2018/01/05 17:21:48 AHRS Info: attempting IMU connection.
2018/01/05 17:21:48 GPS error. Invalid NMEA string: Invalid NMEA message
2018/01/05 17:21:49 MPU9250 mag data overflow
2018/01/05 17:21:49 MPU9250 m4 MSB: 1
2018/01/05 17:21:49 MPU9250 mag data not ready or overflow
2018/01/05 17:21:49 MPU9250 m1 LSB: FE
2018/01/05 17:21:49 GPS detected: u-blox NMEA position message seen.
2018/01/05 17:21:49 setting system time to: '20180105 17:33:11.000 UTC'
2018/01/05 17:21:49 MPU9250 mag data not ready or overflow
2018/01/05 17:21:49 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 Time set from GPS. Current time is 2018-01-05 17:33:11.00073776 +0000 UTC
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 GPS attitude: Error calculating speed regression from UBX position messages
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 GPS attitude: Error calculating speed regression from UBX position messages
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 AHRS Info: Successfully connected MPU9250
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE
2018/01/05 17:33:11 MPU9250 mag data not ready or overflow
2018/01/05 17:33:11 MPU9250 m1 LSB: FE

rex199, 2018-01-08:
I figure this out, you need put the stratus box level and flat and parallel with horizontal line then it works. if you put it vertical from horizontal line, you will keep get msg "MPU9250 mag data not ready or overflow", then you will run out of SD storage space because the log file increases faster.

westphae, 2018-01-09:
Thanks for following up.  This is a very strange circumstance: I've not seen this message repeated so many times before, and I've tested it in just about every orientation.  This obviously shouldn't occur in any orientation.  Does the AHRS behave correctly when you start it vertical?

Aside from the actual magnetometer initialization issue, we should put a hard stop to all logging when the SD card is nearly full, or even better, rotate logs.

mhoran, 2018-01-09:
I've also run out of disk space twice now, seemingly because of stratux.log. The GPS logging is quite chatty, despite having no verbose logging enabled. After a flight last night, I noticed many "No GPS update at ..." messages flooding the log. GPS normally works fine in flight, so I presume this happened on the ground.

It would be great if these logs could be regularly cleared out (although I see the above may be considered another issue -- perhaps I should open a separate one?) I can't see much need to keep the logs around between reboots.

cyoung, 2018-01-09:
Logs are rotated in changes that were committed after v1.4r3.

I've created [v1.4r4](https://github.com/cyoung/stratux/releases/tag/v1.4r4) that includes the changes.

Also we've removed some the unnecessary debug logging that goes into stratux.log.

mhoran, 2018-01-10:
Thanks @cyoung! This looks good to me. I see far fewer duplicate messages in my logs and presume this won't recur. I also appreciate the changes from f28bfff at night.

cyoung, 2018-01-10:
👍 

Will work some more on the goflying module to reduce log output. I think the logrotate addition should keep it under control, but it is only run once on boot.

cyoung, 2018-01-17:
Reopen if the issue reoccurs.

closed 2018-01-17 by cyoung