cyoung/stratux #693: Canadian Weather

SeareyADS, 2018-01-14:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:  There is no weather available on UAT for any Canadian sources.
In the October 2017 issue of Aviation Consumer "Scout vs RXWX" ...
"Young is also developing other interfaces, including a community-sourced ground weather network for Canada."

Can you provide any further information on this initiative and how it might work with the spaced based AIREON (1090).  Is it being done with the knowledge of NavCanada?
NavCanada is currently in the process of consultation with pilot groups in mandating 1090 ADS-B out.  There is no provision for TIS for FIS on the downlink.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2018-01-15:
So this would be done outside of navcanada as they seem uninterested in helping out. 

Work has stalled on this but testing has been done on a system using LORA radios. It works quite well in testing (nearly 100nm range at 3500 feet with a 10 watt transmitter.)  The ground station and receivers use the same hardware which is just a tiny radio that plugs into the GPIO pins. The ground station pulls data from public weather sources.

The biggest hang up is regulatory. Broadcasting needs to be done with licensed hardware and licensed stations. Licensing a Raspberry pi with a lora radio hooked up to a 10 watt power amp might not be cheap.  

If you know anyone familiar with the process at Industry Canada it would be great to get things moving.

SeareyADS, 2018-01-15:
I take it the LORA can be re-tuned to 978, so weather reception would be
via UAT?
   100 nm at 3500 ft sounds like a very useful range and probably
consistent with my UAT receiver in southern Ontario seeing Erie and Buffalo.
    Is this planned to broadcast pre-selected METARs or is there a
negotiation process?

I have the ears of COPA at the moment, trying to re-negotiate a deal for GA
with NavCanada on ADS-B.  Unfortunately COPA has been hung up on the idea
of free ELT coverage with space based AIREON.  However, to do this requires
a top mount antenna with at least 125W at the antenna.  The cost of this
can be enormous in additional to FAA compliant ADS-B. The other downside is
that there is no download capability,   so not even conflicting traffic can
be provided.

 NavCanada maintains an extensive voice comms network to support traffic
within Canadian FIRs but is not using them effectively to distribute
weather or other services.  The CAA in the UK has been conducting trials
with UAT weather and traffic to complement their ADS-B 1090.  See attached.

I do have some experience with Industry Canada and I don't expect the LORA
approach will be viable except for experimentation at a low wattage.
It would be nice to push the process just to show NavCan how easy it can
be.

Lee

SeareyADS, 2018-01-15:
Here is what is on the Industry Canada RF database for 978 MHz

Land-Mobile Digital: 650KF7W   18W
St-Johns Newfoundland

Helno, 2018-01-15:
The LORA stuff we have been doing would be proprietary. LORA Has some tricks up it's sleeve to get really long range with low power. Swithing to straight UAT would mean less range per site but it is entirely possible is access is granted.

This is basically what it happening in the UK. It would be nice if we could convince Nav Canada or transport Canada to just buy the hardware from uavionix and provide it as a public good. It would be nice to just have full compatibility between countries.

It is to bad that COPA is pushing for that. UAT uses way less power and is easy to retrofit. And basically no one has top mount transponder antennas so everyone will have to fork out serious cash for it to be viable.

SeareyADS, 2018-01-16:
I have contacted uAvionix and they are taking some interest.  A high level
exec has committed to responding to the conflicting requirements.
Let's see what happens in next week.
COPA I hope is cooling to the ELT replacement idea after I wrote a letter
to the pres. & sending follow-up to them.  Hopefully we will be pursuing
the CAA path with NC.

Helno, 2018-01-23:
Did you hear anything from either side?

SeareyADS, 2018-01-23:
2nd attempt - attachment was too large....

COPA is a little disturbed by the realization that they're not going to get
a Canadian Alert system from AIREON.

uAvionix took great interest in the topic and provided some information
from the January 15 conference of the Europe.European Aviation Safety
Agency  (EASA).
  Christian Ramsey has been providing me with information from that
conference on "conspicuity".

I will include the paper that he presented and the strategic findings.
  The take away for me was ....
    “Biggest collision risk for GA is other GA”  Bob DARBY, AOPA UK.

He has also provided more information on the FIS-B trials in the UK on
UAT.  You can see some of this in his presentation.
Unfortunately it looks like this is a very short range UAT system.   It
seemed to extend out to only 25 nm.  It has only a 20W transmitter and
seems to be geared to providing only airport weather.  The UK CAA does seem
committed to providing an enroute system.

I have been looking for information on the current FAA ground stations
(GBT).  There seems to be little information about the the details. Yes,
there are block diagrams in the AIM, but how can they claim a 200 nm range
at altitude?  I do get adequate reception in the SW Ontario from the Erie
tower to support the coverage diagrams provided for Google Earth.  If we
persuade NavCanada to put in UAT FIS-B / TIS-B it had better be based upon
a FAA like system.  But I think that ADS-R is a questionable feature for
1090 only.  Be interesting to hear what AOPA (US) has to say about NavCan's
proposal.

cyoung, 2018-01-23:
>He has also provided more information on the FIS-B trials in the UK on
UAT.  You can see some of this in his presentation.
Unfortunately it looks like this is a very short range UAT system.   It
seemed to extend out to only 25 nm.  It has only a 20W transmitter and
seems to be geared to providing only airport weather.  The UK CAA does seem
committed to providing an enroute system.

I got about 75nm @ 3500' transmitting LoRa @ 20W from the roof of my house.

Did you have any information on their antenna, mounting location, etc?

SeareyADS, 2018-01-24:
I got the following document from them which sort of describes the antenna
and location, along with propogation predictions.
     Not sure if that is a stacked dipole as there doesn't seem to be a
groundplane reference.

I'm not sure that this is in the public domain.
Don't think there's anything proprietary in it but don't fire it back to
them please.

DineshkumarGA, 2018-04-14:
Weather transmission seems interesting,
Get in touch,
We are looking for implementation in India.

SeareyADS, 2018-04-14:
You might be interested in the trials and recent changes in the UK.
I received a press release from uAvionix yesterday....

We've also made a presentation on April 11 to Nav Canada to include ADS-B
UAT weather at their remote voice communications locations associated with
each Centre and FSS network.  ATC clearances can't happen without voice
communications on these sites, for now.  The discussion continues !

ICAO has recommended that all aircraft communications be able to
accommodate decreased frequency spacing and digital communications (VDL).
We used this point to our advantage to argue that the remote communications
sites with new radios will need to have 2 way digital/analog comms and
adding ADS-B UAT should not be a big overhead.  Further, we argued that
1090 is too congested and that putting all your surveillance eggs in one
basket is not a great plan.  Further, the planners I hope recognize that
the Aireon satellite system is another single basket solution and adding a
ground based UAT/1090 system will add to the robustness of the ATC system.

Good luck in India.  I worked on the New Delhi Cente radar and comms
upgrade some years ago.  Infrastructure always seemed to be a problem.

Helno, 2019-01-17:
Minor update regarding this. 

http://canada.stratux.me/

Work is progressing on a 1 Watt version of the CC1310 board so that no external AMP will be required.