cyoung/stratux #694: Suggestion: Blinking green light

ghostinstallations, 2018-01-21:
1. Stratux version: 1.14r3
2. Stratux config: Default

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [x] yes
      - [ ] no

    power source: 110 vac adapter

    usb cable:

3. EFB app and version: (ForeFlight v9.5.3 (19754)

    EFB platform: (e.g., iOS 11.2.2 (15C202))

    EFB hardware: (iPad Mini 4)

4. Description of your issue:

Based on the information provided by cyoung in issue #567 the Green LED indicates when Stratux is up and running (typically ~20 sec after power applied). I agree that using this indicator to show when Stratux is operational is the best use of the LED. 

After issuing a shutdown command from the GUI, the LED flashes a few times then remains lit. I am not sure what the flashing indicates, but might I suggest that either the LED is shut off after Stratux is properly shut down?

The desired outcome is to allow the user to know when Stratux is operational and also when it is safe to remove power. Thank you for the consideration.

pp4life, 2018-01-22:
What battery are you using? I bet it's Anker. I have the same problem. There is some "Smart" circuitry in the Anker batteries that is tricked by the Raspberry Pi. As the Pi is shutting down the Anker disconnects. But then the Anker senses a load turns back on rebooting the Pi. I get around this by pulling the power cable right after the green blinking lights stops. 

The green led will stay off to let you know it's OK to pull the power with just about any other battery. I just tried it using a wall power source used to charge a phone.

ghostinstallations, 2018-01-22:
Thanks for the response. As is mentioned in my original post, I am currently testing using the wall power adapter for the RPI3, so I do not believe it is power related.

pp4life, 2018-01-22:
Hmmm, that is really strange. I'm using a RPi2 and v1.4r3-a0d9a4be25.sh (latest DangerZone build). When I'm plugged into an Apple iPhone cube wall wart charger and a random USB mini cable I do not see this behavior. I see the green LED flash a few times, then it stays off. The red LED stays on constantly as it is signifying steady power.

cyoung, 2019-01-31:
Power source and system compatibility issue. Not something we can fix without adding hardware.

closed 2019-01-31 by cyoung