cyoung/stratux #695: Foreflight not displaying AHRS.

Nayajja, 2018-01-23:
[192 168 10 1-logs-stratux log.txt](https://github.com/cyoung/stratux/files/1656351/192.168.10.1-logs-stratux.log.txt)

1. Stratux version:  v1.4r3-a0d9a4be25
2. Stratux config: 

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type:  GPYes

    AHRS
      - [X] yes
      - [ ] no

    power source: PowerBank battery

    usb cable: 

3. EFB app and version: ForeFlight 9.5.3 (19754)

    EFB platform: iOS 11.2.2

    EFB hardware: iPad Air 2

4. Description of your issue:

I have a Stratux that I use with Foreflight. The AHRS works great on the Stratux web page. It used to work with ForeFlight (although with the FF jerkiness). 

Then my SD card failed. I got a new SD card, and re-loaded eveything, with the new versions. I first burned the stratux-v1.4r3-4de2420803.img onto the SD card, and then I updated it with the update-stratux-v1.4r3-a0d9a4be25.sh file.

AHRS is not functional at all on Foreflight now. It works perfectly on 192.168.10.1. GPS is locked and valid, but ForeFlight shows no attitude info.

I am attaching the log at http://192.168.10.1/logs/stratux.log. When I try to retrieve the log at http://192.168.10.1/logs/stratux/ I get "404 page not found".

cyoung, 2018-01-23:
Settings -> WiFi -> click on the "i" in a circle next to the "stratux" network while connected.

What does "IP Address" show?

Helno, 2018-01-23:
Ran into this last week. If the dangerzone patch has been applied and it still is not working remove and reinstall FF/

flyman2456, 2018-01-29:
I'm trying to get my AHRS to work with Foreflight. My AHRS shows up fine when I go to the 192.168.10.1 page but in Foreflight it says "No Attitude information." It displays gps location and gps airspeed just fine but not the AHRS. I have the latest version of FF and the latest stratux release v1.4r4...

Helno, 2018-01-29:
Are you running the Dangerzone firmware? If not no AHRS on FF.

http://updates.stratux.me/builds/dangerzone/

JohnOCFII, 2018-01-29:
> I'm trying to get my AHRS to work with Foreflight. My AHRS shows up fine when I go to the 192.168.10.1 page but in Foreflight it says "No Attitude information." It displays gps location and gps [...]

You need the dangerzone patch.

flyman2456, 2018-01-29:
I am. On FF under devices its showing ADS-B and Stratus 2 as connected. But on maps page its still displaying "No attitude information"

pp4life, 2018-01-29:
I had this happen awhile ago. I ended up doing nothing. 
See Issue #682  
I got sidetracked and came back to it a day later and it was working. Before uninstall/reinstall of FF, I'd try taking it outside and making sure it has a good GPS fix and moving around with it. Like maybe put it on the dash of your car and take it for a spin. Maybe try restarting the Stratux. Mine's been fine ever since I filed that issue #682 and it starting working the next day.

flyman2456, 2018-02-01:
Tried giving it a few days, taking it for a spin in the car, and uninstalled/reinstalled FF. Still no joy..

dbarrington1, 2018-02-02:
Do you have the Dangerzone patch loaded?  If you reinstalled the main program you must reinstall the patch.

flyman2456, 2018-02-02:
Yes the Dangerzone patch is loaded..

ssokol, 2018-02-02:
This sometimes happens to me when I power the Stratux off and back on and the iPad connects to another wifi network. ForeFlight shows both "ADS-B" and "Stratus" in the Devices list. It shows traffic on the map, so it's definitely getting a data stream. Killing the FF app and restarting it usually seems to fix it.

flyman2456, 2018-02-02:
Tried that too.. It shows everything else fine, location, groundspeed, altitude, just no attitude. The attitude shows fine when I open up stratux in the browser. FF is just not getting the attitude info for some reason. Would appreciate any more suggestions, trying to take this flying tomorrow. Thanks.

flyman2456, 2018-02-03:
Took it on a flight and still no attitude information. Flew with another pilot and he had FF on his ipad too. Connected it to his and same thing was happening. Shows AHRS on the stratux page but not on FF...

Nayajja, 2018-02-04:
cyoung wrote: 
Settings -> WiFi -> click on the "i" in a circle next to the "stratux" network while connected.
What does "IP Address" show?

It says:
IP Address: 10.26.36.11
Subnet Mask: 255.255.255.0
Router 10.26.36/1

flyman2456, 2018-02-04:
Mine shows:
IP Address: 10.26.36.10
Subnet Mask: 255.255.255.0
Router: 10.26.36.1
DNS: 4.2.2.2
Search Domains: stratux.local

cyoung, 2018-02-05:
Could you take a screenshot of http://10.26.36.1/#/gps?

flyman2456, 2018-02-05:
[image]

flyman2456, 2018-02-05:
[image]

Nayajja, 2018-02-07:
[image]

flyman2456, 2018-02-19:
Any suggestions?

Helno, 2018-02-19:
There are only two devices that could have a problem. Stratux and your tablet. 

Try removing and reinstalling FF. I had a customer with this issue a few weeks ago and this was what fixed it. 

If it still has trouble then reimage the SD card in your Stratux and then reapply the dangerzone patch.

flyman2456, 2018-02-19:
Okay I'll try reimaging and then reapplying the patch. Already tried uninstalling and reinstalling FF on multiple ipads.

flyman2456, 2018-02-22:
Reimaged my SD card and applied the patch. It is working now! Thank you everybody!

cyoung, 2018-04-03:
Stale issue, likely fixed on both ends.

closed 2018-04-03 by cyoung