cyoung/stratux #696: ADS-B out

GrimShadow, 2018-01-29:
1. Stratux version: 1.4r4
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: Adafruit GPS

    AHRS
      - [ ] yes
      - [x] no

    power source: Battery pack

    usb cable:

3. EFB app and version: Skydemon 3.9.1

    EFB platform: Android 6.0.1

    EFB hardware: Sony Xperia Z3

4. Description of your issue:

Is it possible to add additional hardware or modify the software to provide ADS-B out? In essence I want to try to make a whole package that includes a transponder.

cyoung, 2018-01-29:
Do you mean using Stratux as a GPS source for a transponder? Perhaps.

If you mean using Stratux with some additional hardware to make a DIY transponder, it's likely not happening any time soon.

InstructorPilot, 2018-03-02:
If you are thinking of a portable device, FAA unlikely to approve it. For installed ADS-B OUT, it would need to meet regulatory specs and requirements.

cyoung, 2018-03-02:
Stale issue, closing.

closed 2018-03-02 by cyoung