cyoung/stratux #698: RPi3 and Stratux GPYes

HEWscsi, 2018-02-10:
1. Stratux version:1.4r3-4de2420803
2. Stratux config:????????

    SDR
      - [x ] single
      - [ ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:Stratux GPYes

    AHRS
      - [ ] yes
      - [ ] no

    power source:3A 5V wallwart

    usb cable:microusb

3. EFB app and version: (e.g., WingX Pro7 8.6.2)???

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:I made a mistake apparently purchasing a Stratux GPyes.  My intent is to build a customizable APRS configuration based upon the Raspberry Pi.   I downloaded the img file win32diskimager'd the sdhc card.  Powered up the RPi 2 + and have a command prompt, no documentation, and am puzzled.  The device has a constant blinking green LED even after being up on the Pi outdoors for an hour.  The initial screen says to use stratus --help which is not a command.  The attempt at the address 192.168.101 yields a not found.  Is there any documentation at all for this thing or do I chunk it in the trash and buy a documented GPS?  BTW 41 years in IT with 35 in unix. 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

pp4life, 2018-02-10:
I'm not familiar with exactly what you're trying to do, but I'll give you what I know and hopefully someone else will be able to jump in.
The Stratux project is software for an ADSB-In receiver. It is not software just for the GPYes unit. The GPYes is nothing special as it's just a U-Blox 7 chipset on a custom board with a GPS antenna and some shielding on the underside.

Regarding the web address 192.168.10.1. The Stratux software should make your Pi3 broadcast a WiFi network. Connect to the Stratux network with any device and type in 192.168.10.1 and you should get a webpage with status of the Stratux. However, this doesn't sound like what you're looking for.

Which device has a constant blinking LED? The RPi or the GPYes? The GPYes will blink green when it has a position lock. 

In any case, if you're just trying to use the GPYes for a project, check out ublox's website for software etc.
https://www.u-blox.com/en/evaluation-software-and-tools

HEWscsi, 2018-02-10:
The blinking green is on the GPYes.  The 1 page doc(??) at http://products.stratux.me/gpyes says plug it in and the LED glows red and you are good to go.  Even with the image file from the stratux.me site I get nada when i am on a pc, tablet, or another pi trying to hit the 192.168.10.1 address,  I will check the site that you reference and pray that it is more informative.  It  ain't plug and play.  
ThanksHugh, WB4SLIL

   

https://www.u-blox.com/en/evaluation-software-and-tools—

kdknigga, 2018-02-10:
So, GPYes was designed to work with Stratux, which isn't a APRS but
something else completely.

Plug and play only means something when the involved parties are trying to
play the same game.

So, what is it you're trying to do?

pp4life, 2018-02-10:
Let me take one step back here. It sounds like everything is working as intended for the Stratux project. Does your RPi 2 have the [Edimax WiFi](https://www.amazon.com/gp/product/B003MTTJOY) dongle so it can create a WiFi network? This project has progressed to use the RPi 3 with built in WiFi, but it does still work with an RPi 2 with the Edimax USB dongle (this is the way mine is setup).  

I believe if everything is working properly the RPi should have a solid red LED (power is good) and solid green LED (the system is booted and fully up). The GPYes should have a solid red LED and maybe a blinking green LED depending on which version of GPYes hardware and Stratux software. See issue #659. I currently have version 1.4r4 and I have red and green solid on the RPi and solid red on the GPYes when everything is working.

If you have the Edimax WiFi adapter it should be blinking a blue LED. It's hard to see since the casing is dark grey almost black.

If all that is good, you should be able to go to any WiFi device and see the "Stratux" network. Connect to that and then open 192.168.10.1 in a web browser on that device you connected to the Stratux network.

If something here isn't right, try reimaging the SD card.

This whole project is not ever meant for the user to have to log into the RPi at all, hence there's no documentation other than what you'll see in the code in this repository.

HEWscsi, 2018-02-11:
go bully someone else.  You are not part of the solution since you are not an amateur radio operator.  otherwise you would  know that a gps is used for aprs to report a position  via vhf fm packet otherwise known as ax.25.
     . From: Kris Knigga <notifications@github.com>
 To: cyoung/stratux <stratux@noreply.github.com> 
Cc: HEWscsi <hwardlaw@bellsouth.net>; Author <author@noreply.github.com>
 Sent: Saturday, February 10, 2018 5:08 PM
 Subject: Re: [cyoung/stratux] RPi3 and Stratux GPYes (#698)
   
So, GPYes was designed to work with Stratux, which isn't a APRS but
something else completely.

Plug and play only means something when the involved parties are trying to
play the same game.

So, what is it you're trying to do?

HEWscsi, 2018-02-11:
Thanks for the info.  I will trash the device and buy yet another gps  avoid all the ublox 7/stratus devices since they are so particular.  I am writing to you from my RPI with an 802.11 dongle.  I have already been looking and found one with a serial interface.  I have been working with rs-232 and ttl comm for 41 years.  That and relational databases, operations research have been paying the bills.
Thanks for the help,Hugh WB4SLI

cyoung, 2018-02-11:
@HEWscsi - GPYes should work fine for an APRS setup. I think most of them use gpsd? In any case, our software won't do you any good for APRS.

closed 2018-02-11 by cyoung