cyoung/stratux #700: GPS Track Logs jump around in Foreflight

camaro4life18, 2018-03-26:
1. Stratux version:1.4r4-872839fccc
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ x] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: Portable battery

    usb cable: Micro USB

3. EFB app and version: ForeFlight 9.6 (21535)

    EFB platform: iOS 11.2.6 (15D100)

    EFB hardware: iPad

4. Description of your issue:

I've gone on a few flights recently where I open the track log and it shows my location randomly going back to my home airports FBO. I've tried opening the KML in Google Earth and in the ForeFlight browser, and both show these random jumps. In the screenshot I am attaching I was just flying the pattern, but you can see my flight path would stop going in the direction I was going, then back to the FBO. Almost as if my tablet was switching to the wifi at the FBO and getting the location based on google's wifi location database, but I know it's not connecting to the wifi, as I went on some cross country flights and it jumped back to the same FBO.

Any idea of what's going on or how I can help troubleshoot? I'm a Java developer myself.
[image]

p-easterbrooks, 2018-03-29:
Does the KML reflect the exact same coordinates for the FBO? Also, are you using automatic start/stop with the track logs?

camaro4life18, 2018-03-30:
So, this is strange,
1)  I opened the KML first in the Google Earth app straight from my iPad and saw all the extra lines as shown above. 
2) Uploaded to ForeFlight.com and saw all the same lines. 
3) Downloaded the KML to my desktop and opened it in the Google Earth desktop app...No Lines!

I'm not a KML expert and I'm not sure how to convert the decimals for the Long and Lat, but it all looks pretty close for a 10-15 second period I narrowed down to a problem area. The only thing I did notice, is that the pulling time seems to be about every 1 second on the dot, however during the time period where we are having problems, there's missing data points. There's a 2-3 second gap in location data points.

This makes me think that ForeFlight and the Google Earth app are filling in with their own data points that go back to the start location. 

To answer your question, yes I was using the auto start/stop in ForeFlight. I'm going up for some solo time tomorrow, so I'll try turning off the auto start and manually start the track logs.

Here is a snippet of the KML. Right after 13:49:44 is where I noticed a jump back to start.

`<when>2018-03-19T13:49:36.929Z</when>
        <gx:coord>-85.8916711807251 39.2665958404541 400.8119993907653</gx:coord>
        <when>2018-03-19T13:49:37.928Z</when>
        <gx:coord>-85.89173555374146 39.26696062088013 406.90799938149945</gx:coord>
        <when>2018-03-19T13:49:38.920Z</when>
        <gx:coord>-85.89186429977417 39.26730394363403 411.47999937454995</gx:coord>
        <when>2018-03-19T13:49:39.924Z</when>
        <gx:coord>-85.89203596115112 39.26764726638794 417.5759993652841</gx:coord>
        <when>2018-03-19T13:49:40.926Z</when>
        <gx:coord>-85.89231491088867 39.268012046813965 423.67199935601815</gx:coord>
        <when>2018-03-19T13:49:42.926Z</when>
        <gx:coord>-85.89282989501953 39.26863431930542 440.4359993305368</gx:coord>
        <when>2018-03-19T13:49:44.943Z</when>
        <gx:coord>-85.89345216751099 39.269256591796875 448.05599931895443</gx:coord>
        <when>2018-03-19T13:49:45.939Z</when>
        <gx:coord>-85.89379549026489 39.26955699920654 454.1519993096885</gx:coord>
        <when>2018-03-19T13:49:48.939Z</when>
        <gx:coord>-85.89491128921509 39.27045822143555 463.2959992957896</gx:coord>
        <when>2018-03-19T13:49:49.923Z</when>
        <gx:coord>-85.89529752731323 39.270737171173096 481.5839992679918</gx:coord>
        <when>2018-03-19T13:49:50.939Z</when>
        <gx:coord>-85.89568376541138 39.271037578582764 490.72799925409294</gx:coord>
        <when>2018-03-19T13:49:51.920Z</when>
        <gx:coord>-85.89609146118164 39.27131652832031 499.87199924019404</gx:coord>
        <when>2018-03-19T13:49:52.924Z</when>
        <gx:coord>-85.8964991569519 39.27159547805786 505.9679992309281</gx:coord>
        <when>2018-03-19T13:49:53.921Z</when>
        <gx:coord>-85.89692831039429 39.27187442779541 515.1119992170293</gx:coord>
        <when>2018-03-19T13:49:54.939Z</when>
        <gx:coord>-85.89737892150879 39.27213191986084 524.2559992031304</gx:coord>
        <when>2018-03-19T13:49:55.921Z</when>
        <gx:coord>-85.89782953262329 39.27236795425415 528.8279991961808</gx:coord>
        <when>2018-03-19T13:49:56.924Z</when>`

camaro4life18, 2018-04-02:
Update: I went flying again and did not use the auto start/stop. I did not have any issues with the track logs when I manually started recording my flight.

p-easterbrooks, 2018-04-02:
Awesome. So my line of thinking is that auto-start works by detecting proximity to an airport and GPS movement. My guess is a default starting airport location is used if GPS connection is lost rather than interpolating your position or omitting that data. I've seen similar results with Foreflight when you run a tracklog on a flight sim and pause the flight. 

All this being said, I do not know Foreflight's business logic and I've never had a problem with the internal GPS on my iPad.

camaro4life18, 2018-04-10:
Well, based on what we've found, and the conclusion that it's probably ForeFlight. I'm going to close this ticket and try to open one with ForeFlight.

closed 2018-04-10 by camaro4life18