cyoung/stratux #703: P-alt -- don't send in ownship report if pressurized cabin detected

cyoung, 2018-04-03:
1. Stratux version: v1.4r4
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type:

    AHRS
      - [X] yes
      - [ ] no

    power source: any

    usb cable: any

3. EFB app and version: any

    EFB platform: any

    EFB hardware: any

4. Description of your issue:

Need to detect pressurized cabin and not send pressure altitude to EFB.
Maybe use 14 CFR § 25.841 / § 91.211. GPS alt vs P-alt.

GitUser275, 2018-04-03:
I think this is going down the wrong path.  

Use of ownship GPS altitude as a pressure altitude reference is not appropriate unless there is no source of own-ship pressure altitude.  Even then the traffic relative altitude display is likely to be misleading and an absolute altitude tag would probably be preferable.

Own-ship pressure altitude is included in own-ship squawk.   This altitude is equally valid for pressurized or unpressurized cabins.  May I suggest that this pressure altitude should be received by 1090 SDR and made available for output as own-ship pressure altitude.  

The user should then be given a setup option to use the integrated pressure sensor or own-ship transponder pressure altitude as the output for own-ship pressure altitude.

cyoung, 2018-04-03:
Yes, you're right about that. Ownship altitude and position are derived from ADS-B messages, if available and Mode S hex code is set in Stratux settings.

What I'm looking for here is a method to determine when GPS altitude is the only altitude reference available to the EFB.

ssokol, 2018-04-03:
How difficult would it be to pick up pressure altitude from Mode C? Run
dump1090 with the --modeac option and look for the strongest (by far)
signal. If it's mode C it should be reporting the pressure altitude from
the altimeter or blind encoder.

Axtel4, 2018-04-03:
One of the issue of using Stratux to read your own Mode C altitude is close proximity of the Own Ship transponder, especially with bottom mounted Stratux and transponer antennas.  The high signal level received by Stratux distorts the received transponder pulse stream and causes it to misread the data.  In testing I've done, I've seen it need as much as 60 dB of attenuation for the SDRs to decode correctly.  I think there is a thread on either Reddit or Slack where someone was experiencing this issue when trying to decode their own Mode S output and their solution was to move Stratux about 100 ft away from the transponder antenna in order to get consistent Mode S decodes.

Ergonomicmike, 2018-04-05:
Two things: 1) I've mentioned before somewhere that we have a Monroy ATD-300 which gets out PA info from our Mode C. Occasionally, it needs us to press the IDENT button on our transponder (when we're not squawking 1200) to disambiguate Mode C altitude. (IIRC, Axtel4 has similarly commented before and explained why this is sometimes necessary.)

2) That aside, one brute force option to read the ownship Mode C altitude info might be a third 1090 SDR with the gain set very low in software and/or no antenna connected to it to get a usable (attenuated) signal from ownship.

Ergonomicmike, 2018-04-05:
We had a ZAON PCAS too, and you're correct - it never got confused about Mode C like the Monroy units do/did.

BTW, our ZAON failed years ago. (That's why we bought an ATD-300.) The company had gone under and when I called around to buy a used one, I was told that they had all failed about the same time. Any trick to get it working again?

cyoung, 2018-04-07:
Dual band unit with high gain antennas inside the cockpit (PA-28) with a Narco AT-165, the signal was too strong for the SDRs to pick up. When I took off the 1090 high gain, it picked up ownship with signal at "-9.77 dB".

[image]

Axtel4, 2018-04-07:
I see the age is 13.4s.  I wonder if that is due to the interrogation interval of Statux is missing some of the Mode A/C replies?  A 12 s interval would be reasonable for an en-route interrogation cycle while a 6s interval would be reasonable for TRACON.

GitUser275, 2018-04-07:
"A 12 s interval would be reasonable for an en-route interrogation cycle while a 6s interval would be reasonable for TRACON."

You must live somewhere in the sticks.  My reply light (almost) never goes out flying in the KPHX area.   Don't forget there is a reply for every TCAS interrogation.

Several of the ADS-B Out solutions pick up own ship mode A and mode C from the transponder antenna coax rather than by using a separate antenna.   I think the NavWorx used a clamp on coupler that didn't require disturbing the transponder coax.   Others route the transponder coax through a coupler inside the ADS-B box itself.

Axtel4, 2018-04-07:
The sweep interval from the SSR ground station is either 6 seconds or 12 seconds depending on the airspace type (i.e. enroute v TRACON).  However, the reply interval can be less depending on the number of TCAS interrogations.  Being in a hangar within range of the SSR interrogation and maybe not seeing TCAS interrogations, I was giving a benefit of doubt to the longer interval of 12 seconds.

cyoung, 2018-04-07:
Tested today with the 0-90 dB step attenuator between the antenna and Stratux. When set to the full 90 dB, the results were similar to having the antenna disconnected. With less attenuation, the receiver didn't seem to decode the ownship Mode-C messages. I think the SDR is picking up a signal through the MCX-SMA pigtail. Even so, the signal might be too strong to decode reliably. dump1090 was reporting a number of messages from different squawk codes (0110, etc) with the correct ownship altitude, or the correct squawk code (1200) with no altitude.

I did another set of tests on the ground with the R820T2 LNA gain settings, no attenuator, high gain connected. AGC was not effective for obtaining ownship messages, but it allowed dump1090 to decode messages from distant aircraft. A gain setting of "0.00 dB" via dump1090 allowed it to decode ownship messages reliably, but nothing else (likely because it was in the hangar). Any setting above "0.00 dB" seemed to have issues.