cyoung/stratux #705: gps.go, questional handling of mySituation.AHRSGyroHeading in func makeAHRSGDL90Report()

d-hoke, 2018-04-03:
1. Stratux version: v1.4r4 (5af6d77)
2. Stratux config: irrelevant

4. Description of your issue: 
Following code from main/gps.go:func makeAHRSGDL90Report():
`		if isAHRSInvalidValue(mySituation.AHRSGyroHeading) {
			hdg = roundToInt16(mySituation.AHRSGyroHeading * 10) // TODO westphae: switch to AHRSMagHeading?
		} else {
			hdg = roundToInt16(float64(mySituation.GPSTrueCourse))
		}
`

seems questionable in at least two respects:
1) Appears to be using a value that is found to be invalid
2) in the else branch, the value used is not scaled by 10, as seems to be done with all the other values in that general section of code subject to 
`if isAHRSValid() {
  ...
}
`

cyoung, 2018-04-19:
Not able to investigate now -- @westphae, any thoughts?

westphae, 2018-05-14:
I believe this comment is correct.  Working on validating the result in the gdl90 stream; once validated, will correct, test and push.  Currently this has little to no impact as no EFB's that I know of report the gyro heading.

westphae, 2018-05-29:
Created a PR at https://github.com/cyoung/stratux/pull/720 to fix this.

cyoung, 2018-05-29:
Excellent.

closed 2018-05-29 by cyoung