cyoung/stratux #706: disparity in gdl90 updates vs webui updates

d-hoke, 2018-04-10:
1. Stratux version: v1.4r4 (5af6d77)
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [x] yes
      - [ ] no

    power source: wall/line

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) firefox 38.8.0, proprietary gdl90 consumer

    EFB platform: (e.g., iOS 9.2) (firefox on) slacko puppy linux , (proprietary gdl90 consumer on) Win10Pro 

    EFB hardware: (e.g., iPad Mini 2) (linux on) asus netbook, Win10Pro on udoox86

4. Description of your issue:
The data stream(s) seen via gdl90 sometimes have unexpected mismatches with the data updates displayed in the web browser traffic interface (via firefox on linux)

A1)I have seen multiple updates (say every second for 5-10 seconds, sometimes longer) come through the gdl90 stream for a particular aircraft (identified by 'code'), while that aircraft does not show up at all on the traffic browser page in either the upper section or the lower (basic mode S and No-position messages) during that period of time.
A2)Similar to A1, but the updates are occurring (evidenced by age updates < 1 sec) in browser traffic interface while no updates for that aircraft are being seen on the gdl90 data stream.
B)I have seen updates occur in approximately the same timeframe from both gdl90 stream and in browser interface, but the callsigns will be different, where one side may have an 'N...' sign (which I gather may be internally generated) vs a more recognizeable sign like AALxxxx, DALxxx, UALxxxx etc (AmericanAirLines, DeltaAirLines, UnitedAirLines, etc. ?) - usually this disparity seems to correct itself, with one side or the other acquiring a matching callsign, but this disparity can exist for multiple seconds, what reason would there be for  a delay of more than 1-2 seconds with the web interface and gdl90 updates occurring approximately on a 1 second interval?
C)(This may be a mis-expectation on my part...???) I have seen (multiple) gdl90 data stream updates over a period of seconds (5+) for an aircraft (aren't gdl90 correct for display in the 'upper' browser section?) but that aircraft continues to be displayed for that period of time in the lower section of the browser traffic interface.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

d-hoke, 2018-04-25:
update 
A2 possibilities - 
i) bug in custom gdl90 processor, was not expecting multiple messages in single network transmission
ii) udp packet loss
iii) appears gen_gdl90 has possibility of dropping messages at certain traffic/load levels or if fails to get ping responses over sufficient period of time (and may not work at all if 'ping' disabled on client)

A1 possibilities -
i) as (iii) above, possibility of dropping messages at certain traffic/load levels (BUT I don't believe these limits were close to being reached when this observed) or failure to get ping responses within sufficient period of time (doubtful of this, but possible)

cyoung, 2018-10-10:
>i) bug in custom gdl90 processor, was not expecting multiple messages in single network transmission
Expect multiple GDL90 frames in a single UDP packet.
>iii) appears gen_gdl90 has possibility of dropping messages at certain traffic/load levels or if fails to get ping responses over sufficient period of time (and may not work at all if 'ping' disabled on client)
See here: https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.md#sleep-mode.

Re-open if still an issue.

closed 2018-10-10 by cyoung