cyoung/stratux #709: No UAT Data

lucid2310, 2018-04-23:
1. Stratux version: v1.4r5 (991f48eaa5)
2. Stratux config: 

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ X] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. WebInterface

4. Description of your issue:
I am not receiving any UAT data at all.
[image]

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

kdknigga, 2018-04-23:
Are you in the air or on the ground when you can't receive UAT?

If you're on the ground, this is expected.  The UAT towers are aimed up in
the air, not towards the ground.

cyoung, 2018-04-24:
http://towers.stratux.me/