cyoung/stratux #711: I no longer see a choice to select 1090 or 987 under the setting menu, that part of settings has disappeared . I am running the newest version but it does the same with older versions . I am running a single radio. I use iflygps on my samsung tablet but have looked on an iphone for the settings and on my laptop. Thanks

CalDale, 2018-04-23:
,1. Stratux version:
2. Stratux config: single/dual SDR, GPS yes/no (type?), AHRS yes/no, battery, battery cable.
3. EFB app and version (e.g., WingX Pro7 8.6.2). EFB platform (e.g., iOS 9.2). EFB hardware (e.g., iPad Mini 2).
4. Description of your issue. If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2018-04-24:
They were moved to the "developer" mode. If you really need them, hit the version number on the status page about 10 times in rapid succession. They shouldn't be needed for normal operation.

closed 2018-04-24 by cyoung

CalDale, 2018-04-24:
Thanks. That works. I did need it because I was set on 1090 and only run
one radio.
Appreciate the reply.
Dale

pp4life, 2018-04-24:
CalDale,
    The software is programmed such that if it detects only one SDR, it will automatically set it to 978. You should not need to turn it off. But turning it off manually should not hurt anything.

CalDale, 2018-04-24:
Thanks
How do I now get out of developer mode?
Caldale

pp4life, 2018-04-24:
I think the only way is to reimage the card. Reapplying a .sh update might do it, but I know for sure a reimage will.