cyoung/stratux #713: Can't get Raspberry Pi to boot Stratux (no green LED)

technoluddite, 2018-04-26:
1. Stratux version: tried several from 1.4r1 to 1.4r5
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ ] yes
      - [ X] no
      type:

    AHRS
      - [ X] yes
      - [ ] no

    power source: 2.1A supply

    usb cable: short

4. Description of your issue:

Using a brand new Raspberry Pi 3 B+. Radios and AHRS module not installed at this stage. No GPS module.  It will boot and run Raspbian so the hardware/power/cable appears functional. With Raspbian SD card I get the usual flashing green activity LED during boot. With Stratux SD I do not get a single flash of green, only the red power LED. (Failure seems very early in the initialization) 

Using Etcher to flash + verify the SD card from Windows. Have tried both a 8GB card and a 32GB card. Have tried flashing about 8 times now through various different .img files. (Yes, unzipped) 

I am sure I must be doing something wrong but I am at a total loss - so any help appreciated.

pp4life, 2018-04-26:
The Stratux software does not work with the Raspberry Pi 3 B+. The latest it works with is the Raspberry Pi 3 Model B.

technoluddite, 2018-04-26:
Ok, Thank you - that was easy. Missed that little detail.  
I'll get an older one. Many thanks.

Helno, 2018-04-26:
There is a test image that works with the B+

https://www.reddit.com/r/stratux/comments/85nqif/rpi3b_image/