cyoung/stratux #715: Upside down attitude with AHRS in Foreflight

yasho78, 2018-05-09:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: Battery pack

    usb cable: 

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Foreflight

    EFB platform: (e.g., iOS 9.2) 11.3.1

    EFB hardware: (e.g., iPad Mini 2) iPad Mini 4

4. Description of your issue
I installed the AHRS module and it looks like Foreflight shows the attitude as upside down. The pitch and bank change when I move the box, but the HSI shows a big red 'X'. How do I fix these two issues?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2018-05-10:
When I first tried the native FF AHRS it also appeared upside down. 

What I did to fix it was to ensure the Stratux was level and behaving correctly in the Stratux web interface. Then I went into FF and on the AHRS display clicked on the gear icon. A button then appears that says set straight and level, press that and then verify that the display functions correctly.

ninelima, 2018-09-07:
1) Did you calibrate the gyros?  
2) Give it a try with Kwik EFIS. I have just release the RC-3:
http://members.iinet.net.au/~ninelima/efis/download.html
To set level: Swipe right to left (lift finger) and then swipe left to right on the EFIS.

cyoung, 2018-10-10:
Probably didn't have the orientation set right. I assume you got it @yasho78?

closed 2018-10-10 by cyoung