cyoung/stratux #722: Pi 3 Model B+ support?

rdenisyuk, 2018-06-05:
I don't manage to boot Raspberry PI 3 Model B+ with the latest image, only the red LED is on. Has somebody tried B+ already?

cyoung, 2018-06-05:
No support for the 3B+ yet. There's an image posted on Reddit for experimenting. We haven't found the 3B+ to be better than the 3B in any way, yet.

closed 2018-06-05 by cyoung

jbswindle, 2019-01-17:
I get my Pi from Allied Electronics (warehouse is local, use will call), noticed that 3 B+ is stocked about 6 to 1 over the 3 B and they are the same price ($35) so I just naturally bought, ah, the wrong one.

jocamero, 2019-02-04:
For anyone else looking at old issues, I've been successfully running a 3 B+ and the latest image for about 5 months now without any issues.