cyoung/stratux #723: Unable to compile: go get fails

PepperJo, 2018-06-09:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ X] dual

    GPS
      - [X ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X ] no

    power source: USB charger 5V/2A

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) N/A

    EFB platform: (e.g., iOS 9.2) N/A

    EFB hardware: (e.g., iPad Mini 2) N/A

4. Description of your issue:

I tried to setup a working developer SDcard by following the instructions in the wiki:
https://github.com/cyoung/stratux/wiki/Setting-up-a-working-developer-sdcard

Unfortunately, I was unable to compile stratux:
```
go get -t -d -v ./main ./godump978 ./uatparse ./sensors
main/sdr.go:23:2: local import "../godump978" in non-local package
main/sensors.go:11:2: local import "../goflying/ahrs" in non-local package
main/sensors.go:12:2: local import "../goflying/ahrsweb" in non-local package
main/sensors.go:13:2: local import "../sensors" in non-local package
main/gen_gdl90.go:35:2: local import "../uatparse" in non-local package
sensors/bmp280.go:8:2: local import "../goflying/bmp280" in non-local package
sensors/mpu9250.go:5:2: local import "../goflying/mpu9250" in non-local package
Makefile:17: recipe for target 'xgen_gdl90' failed
make[1]: *** [xgen_gdl90] Error 1
make[1]: Leaving directory '/root/go_path/src/github.com/stratux'
Makefile:14: recipe for target 'all' failed
make: *** [all] Error 2
```

Go version is: 1.10.2 but I also tried with 1.9

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

PepperJo, 2018-06-09:
Ok it looks like go thinks the package is non-local if it is in the proper go_path. Compiling stratux in a different location seems to work. I guess step 16 should be changed in the wiki or did I do something wrong?

cobraPA, 2018-07-12:
I've been going through the same build for a pi zero W here and had some problems with the GO setup as well.  At least with the current version of go, it is recommended to install in /usr/local/go rather than in /root/go as the web instructions currently suggest.  I ran in to some difficulty with how to install and build versus which user to use (pi or root.)  Some additions to those instructs would help, or maybe the go install guidelines have changed?

Pepper, when you say it works in a different location, where did you try and as what user?

I changed my setup to install go in /user/local/go, then used the pi user home for stratux (/user/pi/go/github ...) , however I have not had a successful build yet.  I am seeing:

go build _/home/pi/go/src/github.com/cyoung/stratux/godump978: invalid flag in #cgo CFLAGS: -L/home/pi/go/src/github.com/cyoung/stratux
Makefile:17: recipe for target 'xgen_gdl90' failed
make[1]: *** [xgen_gdl90] Error 1

Will investigate further.

PepperJo, 2018-07-13:
I compiled as root user and I put go in /root as descibed in the wiki. However what does not work is putting the stratux code in /root/go_path/src/github.com/cyoung since it then believes stratux is a non-local package i.e. does not allow non-local imports. To fix this clone stratux in /root

cobraPA, 2018-07-13:
I was wondering about those paths as well.  I suspect /root/go_path/... is not correct?  But I'm new to go, maybe that worked in older versions.  In 1.10 is seems to be an issue that throws errors.

PepperJo, 2018-07-15:
You can set your go path to whatever you want. You just need to set the GOPATH environment variable to point to it.

TamaHobbit, 2018-07-27:
I had the same problem. I don't want to to install go and compile stratux as root, so I installed go the recommended way, and cloned stratux to /home/pi/ instead of /root/.

After following the instructions int he wiki, `make` gave me: invalid flag in #cgo CFLAGS: -L/home/pi/stratux

I fixed this, apparently by adding another export (like the final step of the dev-setup instructions):
export CGO_CFLAGS=-L/home/pi/stratux

alongside the line the final step specifies:
export CGO_CFLAGS_ALLOW=-L/home/pi/stratux

Now compiling works.
Also, I've added these lines to my ~/.profile (or ~/.bashrc) instead of /root/.bashrc.

armeniki, 2018-08-13:
Hi everyone,

I'm afraid I am having the same issue and have been trying for a few days now to get a successful build.

> pi@raspberrypi:~/stratux $ go env
> GOARCH="arm"
> GOBIN=""
> GOCACHE="/home/pi/.cache/go-build"
> GOEXE=""
> GOHOSTARCH="arm"
> GOHOSTOS="linux"
> GOOS="linux"
> GOPATH="/home/pi/stratux"
> GORACE=""
> GOROOT="/usr/local/go"
> GOTMPDIR=""
> GOTOOLDIR="/usr/local/go/pkg/tool/linux_arm"
> GCCGO="/usr/bin/gccgo"
> GOARM="6"
> CC="gcc"
> CXX="g++"
> CGO_ENABLED="1"
> CGO_CFLAGS="-L/home/pi/stratux"
> CGO_CPPFLAGS=""
> CGO_CXXFLAGS="-g -O2"
> CGO_FFLAGS="-g -O2"
> CGO_LDFLAGS="-g -O2"
> PKG_CONFIG="pkg-config"
> GOGCCFLAGS="-fPIC -marm -pthread -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build246329765=/tmp/go-build -gno-record-gcc-switches"

When I run make, I get the following error:

> Makefile:17: recipe for target 'xgen_gdl90' failed
> make[1]: *** [xgen_gdl90] Error 2
> make[1]: Leaving directory '/home/pi/stratux'
> Makefile:14: recipe for target 'all' failed
> make: *** [all] Error 2
 
I'm really frustrated as I've been searching for various solutions online but nothing seems to work.

Hopefully someone can help! 

Cheers.