cyoung/stratux #724: Raspberry Pi 3 B+ will not boot with Stratux SD card

youcangetme, 2018-06-11:
1. Stratux version: v1.4r5
2. Stratux config:

    SDR
      - [ ] single
      - [ X] dual

    GPS
      - [ X] yes
      - [ ] no
      type:

    AHRS
      - [ X] yes
      - [ ] no

    power source: bench power supply tuned for 3 B+, 2.5 amps @ 5.1v

    usb cable: yup, got one...

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Never even get there

    EFB platform: (e.g., iOS 9.2) Android O

    EFB hardware: (e.g., iPad Mini 2) Everything...

4. Description of your issue:

I looked around and didn't see any bugs yet for the Raspberry PI 3 B+ so I'm filing a new ticket;

This is a new install. Brand new Raspberry PI 3 B+, SD card etc. The Pi B+ will not boot with the Stratux SD card. Remove the card and insert the card into another older raspberry pi and it boots. Made sure the new Pi wasn't defective by loading new OS sd card and it boots just fine.

Symptoms; Red light stays on and the rainbow screen is constant.

There are no logs.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2018-06-11:
Pi3B+ isn't supported. Stick with the Pi3B.

closed 2018-06-11 by cyoung

youcangetme, 2018-06-11:
Interesting, no horizon for support?

cyoung, 2018-06-11:
No. There's no benefit at this point and the power consumption is quite a bit higher.

youcangetme, 2018-06-11:
Ha! Raspberry Pi even in the B version will brown out with 2.4 amps @ 5 volts. I see the red light flashing on the thing all the time. You have to adjust the output to 5.1 volts to stop it.

cyoung, 2018-06-11:
Sounds like your microUSB cord is the issue.

youcangetme, 2018-06-12:
:-) no issues with any of the equipment accept the unsupported B+. Have a great night.

gonzalu, 2018-11-07:
Thanks for this thread. Was pulling my hair out. 

Is this fact that the B+ is not supported written anywhere on the main site or otherwise? Not a big deal as I have older Pi boards at my disposal but would be good for a new person to know. 

Cheers!