cyoung/stratux #725: Jumping GPS position in ForeFlight

CraXgt, 2018-06-17:
1. Stratux version: v1.4r5
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: VK-162

    AHRS
      - [X] yes
      - [ ] no

    power source: 2.1A 12V USB Power adapter

    usb cable: ordinary 20cm USB cable

3. EFB app and version: ForeFlight 10.1

    EFB platform: iOS 11.4

    EFB hardware: iPad Mini 2

4. Description of your issue:

Ownship position jumps every second to a position way off and back. This only occurs when autoscroll of map is enabled. (see video here https://www.dropbox.com/s/4vtqy3mbw947aqd/ff-jumpy-gps.mov?dl=0) 
Looks like the position messages contain a false one or heartbeat message that FF doesn't like. Is this a GPS-Mouse issue or a Stratux one? What about the FFCrippled option? I can't find a documentation of what its' doing. I found no difference with a VK-172 I tested. Hope someone has an idea.
Thanks for your help guys.

Ergonomicmike, 2018-06-24:
Interesting. I have been telling the guys in Slack about a similar phenomenon in iFly, that the Map in iFly sometimes jumps around. In iFly it doesn't occur every second. But it occurs often enough that it sounds like iFly is sampling "bad" data every now and then. (i.e. aliasing.)

CraXgt, 2018-06-25:
Been trying to debug this problem but wasn't able to find anything wrong with those messages. Contacted FF developers to resolve this issue together. Let's see - I'll keep you posted.

0x74-0x62, 2018-06-25:
I have been in touch with the ForeFlight connectivity developers over the last days. After some debugging, it turned out that this problem seems to be caused by some ForeFlight-internal data handling. The data that is sent by Stratux seems to be correct. This problem will be fixed in a future ForeFlight release.