cyoung/stratux #726: Foreflight not showing attitude information w/ Stratux

stevendt1, 2018-06-30:
1. Stratux version:
v1.4r4
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ x] yes
      - [ ] no
      type:

    AHRS
      - [ x] yes
      - [ ] no

    power source: Battery pack

    usb cable: Yes

4. Description of your issue:
When using my stratux, foreflight does not show attitude information and says: no attitude information. I'm updated to the latest build, and my ipad is currently running IOS12. It wouldn't work on my previous IOS's either. Please help!

pp4life, 2018-06-30:
It's right there in your problem report. You are using software version1.4r4. You have to be on 1.4r5 or above to make it work properly with Foreflight. You didn't say what version of Foreflight you have, but it needs to be Foreflight 10 or above.

stevendt1, 2018-06-30:
Where can I find that download?

JohnOCFII, 2018-06-30:
> Where can I find that download?

https://github.com/cyoung/stratux/releases/tag/v1.4r5

closed 2018-09-08 by Helno