cyoung/stratux #727: Random Reboots w/ AHRS module

danielschonfeld, 2018-07-10:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: External 

    AHRS
      - [X] yes
      - [ ] no

    power source: Power socket or battery

4. Description of your issue:

Once I install the AHRS stratux module (MPU+BMP) the device experiences random reboots a few seconds after it boots up.  Usually associated with movement of the device.  

The raspberry pi board I have is an older one - "Rasperry Pi Model B+ V1.2" - Not sure if thats the source of the problem.

Whenever I remove the AHRS module, everything works fine without reboots.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

pp4life, 2018-07-11:
Is the red LED staying on solid the entire time? With the added current draw of the AHRS board it might be enough to drop the voltage low enough to make the RPi reboot. The red LED should be solid on the entire time. The Green LED should be solid the entire time once the system boots. If the red LED is going out, the usual suspect is a cheap, low gauge power cable (regardless of how good your power supply is).

danielschonfeld, 2018-07-13:
@pp4life i think you're right.  A different cable does seem to solve the problem.

closed 2018-07-13 by danielschonfeld