cyoung/stratux #728: Display CPU temperature range over the operational period

rvernica, 2018-07-19:
This is a feature request. On the home page, it would be nice to have range of the observed CPU temperature range next to the current temperature. That is, display the minimum and maximum temperature observed since the unit was started.

[This](https://www.raspberrypi.org/help/faqs/#performanceOperatingTemperature) is what I found regarding the Rasberry Pi operating temperature specification.

cyoung, 2018-07-30:
Sounds like that's an easy addition -- we can add it the help page.

rvernica, 2018-07-30:
Sorry for being confusing. The intention was to display the actual minimum and maximum observed temperatures since starting up, not just the specification range. So, for example, display:
```
CPU Temperature: 40.1 [35.2-50.8] C
```
Where `40.1` is the current temperature while `35.2` and `50.8` are the minimum and maximum temperatures observed since the last reboot, respectively.

Additionally, a green font can be used if the number is within the normal operating range and a red font  if it is outside of the operating range.

cyoung, 2018-07-30:
Ah, right. I can't add it now but if it is added, would prefer that it be visible in "DeveloperMode" only. To date I haven't heard of a single case of a RPi overheating in a Stratux setup, so I'd question the value of having a min/max in there.

cyoung, 2019-01-31:
I've added the backend features for this and added it to the `stratux.log` status message:

```
2019/01/31 16:36:29  - CPUTemp=45.08 [44.55 - 48.31] deg C, MemStats.Alloc=1.6 MB, MemStats.Sys=5.4 MB, totalNetworkMessagesSent=30
```

[image]

You'll need to force refresh the webui to see it.

Anyone want to add colors or change the display?

scubadogrick, 2019-01-31:
Remember time spent flying is not subtracted from your life!

Rick Herbert

commit 8d9fdd31 (2018-07-30): Add info to main status help page.

Fixes #728.
closed 2019-07-02 by Helno