cyoung/stratux #729: Config Stratux for AirNavigatonPro

ohne80, 2018-07-28:
1. Stratux version: 1.4r5
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [x ] yes
      - [ ] no
      type: exrern VK162

    AHRS
      - [x ] yes
      - [ ] no

    power source: powerbank 10.000 / 5.0V, 2,1A

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
Air Navigation Pro
    EFB platform: (e.g., iOS 9.2)
iOS 11.4.1
    EFB hardware: (e.g., iPad Mini 2)
iPhone 7s and ipad Mini 4
4. Description of your issue:
i need help to config the stratux for Air Navigation Pro. I tried to follow the instruction from this site (0sander/stratux_air_nav_pro). but i think i am not clever enough.
in standard config it works, i get information in flight box utility

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

ohne80, 2018-07-28:
i changed the IP and so on, but i cant change the port!? The stratux.conf is completly empty

cyoung, 2018-10-10:
@ohne80 - change a setting in the webui -- any setting. stratux.conf will then be saved and you can edit it.

cyoung, 2018-10-10:
Here's the tutorial referenced: https://github.com/0sander/stratux_air_nav_pro.

I've sent them an email. This should be changed on the EFB side, since they would be one of the few that is not using port 4000.

Will circle back later, if we don't hear anything back in a while then will close this.