cyoung/stratux #730: Stratux restarting

mattphp1, 2018-07-29:
1. Stratux version: 1.4r5
[Safari - Jul 28, 2018 at 8:11 PM.pdf](https://github.com/cyoung/stratux/files/2238416/Safari.-.Jul.28.2018.at.8.11.PM.pdf)

2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: kmashi k-mp816

    usb cable: standard

3. EFB app and version: (e.g., WingX Pro7 8.6.2) ForeFlight 9.6

    EFB platform: (e.g., iOS 9.2) 11.4

    EFB hardware: (e.g., iPad Mini 2) iPad mini 2

4. Description of your issue:  stratux device keeps restarting no other issues with functionality. I have attached the logs
[text.txt](https://github.com/cyoung/stratux/files/2238413/text.txt)

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

wa9als, 2018-07-29:
Are you sure it’s not intermittent power? USB C plugs known for wearing down and being loose, and female connection known for lint and looseness. I flew almost 3 hrs yesterday and was frustrated by stratux. Then I found it was only the cable being worn from multiple uses with an earlier phone.

mattphp1, 2018-07-30:
I cleaned out the usb c receptacle on the pi and tried multiple cords and power supplies. It won’t stay on steadily. Definitely frustrating.

cyoung, 2018-07-30:
Does the red light turn off when it restarts?

If so, try another cable. If that doesn't work, the Raspberry Pi needs to be replaced.

mattphp1, 2018-07-31:
The red light doesn’t turn off when it restarts. It stays solid for the most part but will start flickering intermittently regardless of the cord I have attached. I didn’t have any issues with it until the most recent release.  Any other thoughts before I rip it apart to replace the Raspberry Pi?

mattphp1, 2018-08-16:
Got a new pi board and having the same issues. Won’t stay on for more than 6-45 seconds before it restarts. Any ideas?

armeniki, 2018-08-16:
@mattphp1 This definitely sounds like a power supply issue.  Have you used another one to confirm?

mattphp1, 2018-08-16:
I’ve tried band new usb cables and plugged directly into wall power. It still cycles between 6 and 90 seconds. Even when plugged into the wall the red light doesn’t stay on on the RPi. It comes on initially but once the software runs it never comes back on. Any other ideas?

Nokomis449, 2018-08-16:
A blinking red light light indicates a power problem, period. If it's not the power supply, then it's the cable. I know that's hard to believe, but USB cables are made to charge devices, not power them. Therefore, the majority are made with thinner wires. Until you buy a quality cable that is made with wires thick enough to carry the load required by the pi, you are going to get a blinking red light. Order an Anker or other quality cable from Amazon. Until you fix the blinking red light, you are not going to fix your problem.  I went through a whole drawer full of cables before finding one that would work. That's when I ordered Anker cables, and have not had a power problem since.  As Hans and Franz used to say, listen to us now and hear us later.

mattphp1, 2018-08-16:
I’ll try a different one again. The strange thing is I’ve used this for the past two years and didn’t have any issues until recently. Was there a change in the software that required a higher power pull?

Nokomis449, 2018-08-16:
I can't speak to recent changes. One thing you can try is to remove your SDR and GPS and see if you still get a blink. If not, then the power is inadequate. If it still blinks though, you haven't really proven anything.

mattphp1, 2018-08-16:
With the new RPi board the red light doesn’t even flicker. Only comes on right at start-up and then never comes back on.

mattphp1, 2018-08-16:
I’ll try the SDR and GPS thing today after work and report back. I bought an Anker cable too. Just for posterity sake.

armeniki, 2018-08-16:
@mattphp1  You mention new cables but have you tried a different power supply?  Also, switched-mode power supplies that are commonly used in those supplies are not perfect and they can fail overtime, producing less power than when new.

braddward, 2018-08-16:
If you haven't tried a new flash card, I'd make a fresh one on a new card
and try that.  It sounds like you've replaced about everything else.

Do the logs show anything?

Nokomis449, 2018-08-16:
You can also connect your Pi to a TV via HDMI and watch the boot up process.  Maybe there's a clue there...

mattphp1, 2018-08-18:
I got a new anker cable and power supply. The red light now stays on (getting enough power). The stratux software keeps resetting at the same intervals. It will stay on for anywhere between six and 45 seconds or so before it restarts. Any other ideas here? I’ve literally replaced everything short of the gps/ahrs chip and receiver chips.

mattphp1, 2018-08-18:
I’ve tried a brand new micro sd chip as well.

armeniki, 2018-08-18:
Please provide the output specs for your new power supply.

Nokomis449, 2018-08-18:
Have you connected to an HDMI monitor and watched the output as it boots up?  Any clues there?

Now that you have adequate power, I'd re-flash the SD card in case the low-power issue caused a corruption.

Also, try booting up with no SDR's or GPS connected and see if it can complete the boot.  If successful, you'll have a solid red and solid green light after a minute or two.

Also, and I forget what all has been mentioned so at the risk of mentioning it again, the Stratux image does NOT run on the latest Pi board.

armeniki, 2018-08-18:
@Nokomis449  I don't think we've established that there's adequate power just yet as mattphp1 has not provided any info on the new supply he purchased.

@mattphp1 You should also note that just because the red LED stays on doesn't necessarily mean there's adequate power being received.  Let us know about your new supply specs.

Cheers

Nokomis449, 2018-08-18:
Steady red light indicates adequate power.  Here's a quote from a rather lengthy and informative post by the Raspberry Pi people themselves:

Later models of the Pi (from the B+ onwards) have a new under voltage detector. This turns the power LED off when the input voltage drops below 4.65V. In this case, you can be reasonably sure that, if your power LED is illuminated, your power supply is sufficient. The only exception is if you are using USB devices which cannot handle a drop to 4.65V. Try unplugging all USB devices during boot to see if this helps: USB ports on later models of the Pi are hot-plug-able, so you can plug USB devices in after booting without crashing the software.

https://www.raspberrypi.org/forums/viewtopic.php?p=437084#p437084

armeniki, 2018-08-18:
@Nokomis449  That's good to know, although it says "reasonably sure" whatever that means.

Let's see what mattphp1 comes back with... I'm really curious now!

mattphp1, 2018-08-18:
So, I switched back to the RPi 2 board and still had the same rebooting issue. The red light stays illuminated. No lightning bolt on the screen when I connect the hdmi. What I did discover though is that when I disconnect the leads for the AHRS from the pins on the RPi board the issue stops. I have the RY835AI GPS chipset. Like I said at the beginning of this thread; it seems to have started having issues only after a certain build was released. Maybe R4? Thoughts at this point?

Nokomis449, 2018-08-18:
RY835AI support was discontinued a long time ago.  Switch to a Stratux AHRS board and a USB GPS.

mattphp1, 2018-08-18:
Okay, must’ve missed that one. I have the RY835AI plugged in via USB port and was using two jumpers for the ahrs. Can I still used the 835 for just GPS via USB? It still seems to pull the coordinates and talk to stratux just fine. I don’t want to have to change cases is why I ask.

Nokomis449, 2018-08-18:
I think that'll work, and you can optionally add the $15ish Stratux AHRS if you want.  Maybe someone smarter will chime in and confirm.  Meanwhile, you can close this issue.

mattphp1, 2018-08-18:
Got it, I already ordered the Stratux AHRS.  Thanks for helping me work through this, guys!

Safe flying!

closed 2018-08-18 by mattphp1