cyoung/stratux #731: Fan not working at all

tobo07, 2018-07-30:
1. Stratux version: PepperJo‘s Image
https://github.com/PepperJo/stratux/commit/ee303aa1e693a201b122918c7eac9c9ff4ca386b
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes, on AHRS Board (FlightBox)
      - [ ] no
      type:

    AHRS
      - [ x] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) EasyVFR

    EFB platform: (e.g., iOS 9.2) various android and iOS

    EFB hardware: (e.g., iPad Mini 2) various

4. Description of your issue: I am using a FlightBox with additional sdr for Flarm reception and usb rs232 Converter cable programmed to Stratux Serialout. Fan and the 3 leds on the AHRS board are working ok with latest FlightBox beta version 1.4.5. After now having loaded the above mentioned Flarm receiver image everything works ok but the Fan remains dead and the 3 led lights on the AHRS board are dark as well. External power seems ok as red and green led on main board are on. Temp is going up to 70 deg C but Fan does not run. I tried several reboots but no success. As I am new to rasp, ssh and github I have no idea what‘s going on. any help welcome
Thanks
Thomas

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

ssokol, 2018-07-30:
The fan and the lights on the FlightBox AHRS board are controlled by a process called "boardctl" which is started automatically by systemd. That daemon uses another called "pi-blaster" to manage the PWM on the fan. I would start by making sure both of those are running.

cyoung, 2018-07-30:
Sounds like the software was upgraded to a FLARM-enabled version from the stock FlightBox config, so maybe boardctl wasn't enabled/installed.

tobo07, 2018-07-31:
Thanks a lot for the quick replies.

Does that mean whatever stratux version I will use on a stock FlightBox that I will allways have to add boardctl and pi-blaster? Would that be enough or do I have to change anything else in order to avoid any clashes? Will FlightBox images once Support the flarm reception?

tobo07, 2018-08-03:
Installed boardctl and Fan and led working. Thanks for your advise

closed 2018-08-03 by tobo07