cyoung/stratux #733: Stratux reliably drops out in a Manhattan tour

robert-jm, 2018-08-02:
1. Stratux version: v1.4r5
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: External USB puck

    AHRS
      - [ ] yes
      - [X] no

    power source: Previously-recommended EasyAcc 6Ah battery w/built-in USB-micro

    usb cable: (see above)

3. EFB app and version: Latest ForeFlight

    EFB platform: Latest iOS

    EFB hardware: iPad Mini 2

4. Description of your issue:

My iPad reliably loses its connection to Stratux when I do a city tour, which involves circling the island of Manhattan at 1500-2000'. I thought the unit was simply flaky, since it has been in the past. But I did two flights in the same day last time - a ~100nmi round-trip, during which the Stratux worked 100% perfectly, and the city tour, during which it disconnected.

I have the ForeFlight track log, which I've attached. You can see when it dropped out by when the vertical lines representing the points disappear. It was disconnected the entire way south from about the Harlem River until the southern tip of Manhattan (missing a turn over Central Park and south down the East River), good around the Statue of Liberty, then disconnecting again between the southern tip and the Harlem River going back northbound.

I didn't think to enable the replay logs in advance (I'll do that next time), but I did open the webapp on the northbound return and the uptime was 50 minutes (since before engine start), so the unit hadn't rebooted. I was having trouble getting the webapp page to load, though it's a high workload time so I didn't notice specifically if that correlated with the outage.

- Is it possible that the device is overloaded with traffic in the vicinity? Is this known to happen? I have the high-gain antennas, so I pick up tons of traffic in the area from the three major airports and all the choppers. It runs hot, but I have the cooling fan and it never gets what I'd consider dangerously hot for silicon.
- Are there any known iPad performance requirements in the face of high traffic? The Mini 2 is showing its age; is it possible it just couldn't keep up?
- Is it possible that some government security ... thing could jam the Stratux? It seems roughly centered on Trump Tower... I lose both traffic and GPS position, so it doesn't seem like a GPS jammer as that shouldn't affect traffic reception, right? (Foreflight considers the Stratux to have disconnected during the outage.)

Mostly I'm asking if there are any known explanations for this behavior, specifically if there's any known limits on nearby traffic density due to the SDRs or processing capability. I've never done the flight with a Stratus device, I could try that and see if it does better during that portion of the flight. Presumably if it's an overload issue, the non-commodity-hardware in a Stratus would do better. But maybe it's not that at all and I can try something else.

(I've been really impressed with the reliability recently. I had serious disconnect issues until the April build, resolved by minimizing and re-opening Foreflight, but since then - with the exception of this flight - it's been quite reliable. Thanks for all the hard work!)

[kcdw-hudson-stratux.kml.zip](https://github.com/cyoung/stratux/files/2251888/kcdw-hudson-stratux.kml.zip)

pasniak, 2018-08-02:
I am in the middle of the lines without GPS and (on the roof) I do get ~30 radius of ADS-B traffic (as of 7/31 7pm - when I was testing the high-gain and the white box) and GPS. I kept it running for 5min or so.

Helno, 2018-08-02:
I doubt it is being overloaded by traffic. I had to turn off traffic in Foreflight during the arrival to Oshkosh since Foreflight was getting really bogged down.

Helno, 2019-01-17:
Has this continued to be an issue?

closed 2019-01-25 by Helno

robert-jm, 2019-07-16:
It had been, but I just did a city tour on Saturday and it was fine, though I didn't do the East River that time. 

A few things have changed:
- New iPad Mini instead of v2 from ~2014. (But I seem to recall checking the Stratux logs the last time this happened and seeing evidence it was on the Stratux, not iPad.)
- New low-power ASIC-based UAT radio, and updated Stratux image for compatibility.
- Quiet airspace due to an imminent Yankees TFR.

I'll make sure logging is enabled next time in case it happens again, and I'll re-open if so.