cyoung/stratux #735: NMEA serial output

tobo07, 2018-08-12:
1. Stratux version: v1.4r5
2. Stratux config: OFS hardware

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes, internal on AHRS board
      - [ ] no
      type:

    AHRS
      - [ x] yes
      - [ ] no

    power source:

    usb cable: CP2102

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue: No NMEA serial output.
As of v1.4r4 Stratux has been announced to support the output of NMEA formatted GPS data via the serial output. I am using a CP2102 USB serial converter programmed to "Stratux Serialout" but that only seems to deliver ADS-B out data for the use in an EFIS (at least I can see some messages with ads-b track identifiers on my PC).
My question is now: 
What needs to be set in stratux to output the NMEA (GPRMC) messages?
Should I be able to read those messages also in static conditions (not flying)?
Do I need a different cable and what type and configuration would it be? (amazon??)

Would be nice if someone could answer!!

Thanks a lot 
Thomas

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2018-08-14:
I don't recall NMEA output over serial on the main Stratux image. 

You might be thinking of the work @ssokol did on his modified flightbox code. 

http://support.openflightsolutions.com/t/beta-feature-serial-output/80/15

tobo07, 2018-08-15:
@Helno Thanks for the link, I will sort it out.

closed 2018-08-15 by tobo07

royerm54, 2019-09-11:
@tobo07  were you able to sove/get around the NMEA serial question????  I'm having the same question and situation as you as I'm trying to connect it to an EZ Pilot autopilot
Regards..