cyoung/stratux #737: Profile/optimize traffic page

cyoung, 2018-08-21:
https://github.com/cyoung/stratux/blob/master/web/plates/js/traffic.js

Locks up on some devices with a large traffic target list.

JCBird1012, 2018-09-10:
Is there any further info on reproducing this issue? It may be something that I want to tackle if I can reproduce it somewhat consistently.

cyoung, 2018-10-10:
@JCBird1012 - it seems to mostly show up on older Android devices. Issue reported on:

LG G6. Android 7.0. Chrome.

Samsung Galaxy Tab A.  Android 7.1.1. Chrome.

You can probably get the code to lock up on any platform if you un-comment this and re-compile:

https://github.com/cyoung/stratux/blob/7340719ae3ebd4c3dd6c8991137bc4bb2f3b1c72/main/gen_gdl90.go#L769-L783

Increase `numTargets` until it crashes or locks.

OEZWONKP, 2019-02-17:
Seems like i have the same issue, will report soonest

OEZWONKP, 2019-02-17:
Traffic Page locks consistently on all platforms, chrome(ium), IE on W7, Firefox, Vivaldi. Happens when target list gets large (and scrolls to next page i think) Interestingly, if you deactivate the wifi on the client side the page  refreshes correctly just before the disconnect! It's making my HW unusable so id appreciate some work here.

crhiles, 2019-06-30:
I am seeing the traffic page lock up quite frequently.  It does not recover for me after the amount of traffic decreases.  I cannot go to any other pages on the Stratux interface.  
Some screenshots showing the amount of traffic.  This was at FL430 just south of Chicago.
iPad IOS 12.3.2
[image]
[image]