cyoung/stratux #739: Multiple Mode S Exclusions?

tippmar, 2018-08-21:
I'm in a flying club with several planes, all with Mode S transponders. I rotate among the planes regularly and constantly fight the "TIS-B Shadow" issue in Stratus / Foreflight.

I know I can add a single Mode S hex code on the Stratux setting page to exclude that ID and prevent the TIS-B "shadow" issue - but it would be nice to enter a list of Mode S code values, one for each plane in the club's fleet, so that I don't have to always remember to change that value when I switch planes.

I'm willing to take the risk of not seeing traffic info for another club plane that might be flying at the same time as me - in my case, it's a remote possibility. 

Can this be added as a feature to Stratux?

JohnOCFII, 2018-08-21:
I'd love this too  I instruct in many planes - but I think someone is going to have to volunteer to add this feature. :)

tippmar, 2018-08-21:
Maybe a good opportunity for me to get my feet wet in Golang...

JohnOCFII, 2018-08-21:
I was thinking a drop-down to choose the plane I'm flying, but yeah, an easier solution would be to just build a list and ignore those IDs at all times.

jonkman, 2018-08-21:
How about having it learn it's ship's ID by listening for the ID closest to it within the first 60 seconds after boot and ignore that. Perhaps verify by looking again in 15 minutes to make sure that ID is still within a few hundred feet of the stratux.

tippmar, 2018-08-23:
Thanks, @maehtricks  - that seems like a reasonable approach. I'm playing with modifying the "settings" page now, with the intent of storing the list of Mode S ID / Callsigns in browser local storage. Not a perfect solution - it would be better to send the list of Mode S IDs to Stratux to store internally on the device, but it's a pretty non-invasive change to do it on the browser.

peepsnet, 2019-10-24:
I would suggest the dropdown with multiple planes and one must be selected. 

The problem with multiple values at once is the chance of you not getting alerted of traffic from another plane in your settings out in a practice area or in the pattern and actually colliding.

pp4life, 2019-10-24:
> 
> 
> I would suggest the dropdown with multiple planes and one must be selected.
> 
> The problem with multiple values at once is the chance of you not getting alerted of traffic from another plane in your settings out in a practice area or in the pattern and actually colliding.
> 
> Could you imagine Embry Riddle or UND... You add all the planes then run into your buddy in the practice area

This exactly. There are 10 planes in our Aero Club. I fly all but 1 of them and several are often clustered in the practice area.

Koudy232, 2019-10-24:
If you do not use this feature you can just file only one hexcode. I am having onboard usually two devices, which shall not be traced (transponder and OGN/FLARM) so for me it is big benefit. It has been already incorporated to European version (https://github.com/b3nn0/stratux) and works well.