cyoung/stratux #741: Devel/Flarm_Receiver with NMEA serial out

tobo07, 2018-08-31:
1. Stratux version: 1.4r5 with /devel/flarm_receiver
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual 1090 and 868 (FLARM)

    GPS
      - [ x] yes on AHERS board (Flightbox)
      - [ ] no
      type:

    AHRS
      - [ x] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
easyvfr
    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

I am running Stratux 1.4r5 with the devel/flarm_receiver based on pepperJo's build which runs apart from some FLARM decoder restarts ok. Now I tried to merge the nmea-flarm function (from 1.3r2 ??) (see also #489  and #571 ) into it to achieve the output of the NMEA 0183 sentences (GPRMC, etc) which would be transferred via a CP2102 usb2serial converter into the transponder. After the adaptation of the Makefile and mainly the flarm-nmea.go file and some others go files to 1.4r5 I achieved to get the build running but with strange behaviours. E.g. the build never gets to the call of printstats or the cputempmonitor call in gen_gdl90.go hence the shown temp at settings page always remains zero and therefore no fan coming up. (this is only one symptom and I think I have a funamental mistake here which I am not able to solve.
I tried for some days now to get it fixed but I am new to Go and would like to ask if someone can help and build a version based on 1.4r5 plus devel/flarm_receiver and merging in the nmea-flarm branch or any other working serial output of NMEA fomatted messages?

Thanks a lot
Thomas

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

maehtricks, 2018-09-02:
Sounds like your code is blocking and not running concurrently.
You'll probably have to alter your call in gen_gdl90.go to this one: `go tcpNMEAListener()`

tobo07, 2018-09-08:
Thanks @maehtricks,
Currently I have no access to my rpi but will try when I'm back from hols. 
Thomas

cyoung, 2018-10-10:
Linking the code in question for reference:

https://github.com/PepperJo/stratux/tree/devel/flarm_receiver

Couldn't see any obvious issue here. @PepperJo?

PepperJo, 2018-10-11:
I don't know of any of such issues but that might be related to the merge with the nmea-flarm branch.

closed 2018-11-26 by cyoung