cyoung/stratux #743: AHRSTurnRate always shows: 3276.7

ninelima, 2018-09-05:
1. Stratux version: v1.4r5
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: USB
    usb cable: yes

3. EFB app and version: Kwik EFIS rel 22

    EFB platform: Android 4.4.2

    EFB hardware: Nexus 7 Tablet and Samsung Galaxy S2

4. Description of your issue:

 AHRSTurnRate always shows: 3276.7

`                    //----------------------------------------------
                    // use the Http
                    try {
                        // Situation
                        String situation = getSituation();

                        JSONObject jObject;
                        jObject = new JSONObject(situation);

                        GPSSatellites = jObject.getInt("GPSSatellites");
                        GPSSatellitesTracked = jObject.getInt("GPSSatellitesTracked");
                        GPSSatellitesSeen = jObject.getInt("GPSSatellitesSeen");

                        AHRSPitch = jObject.getDouble("AHRSPitch");
                        AHRSRoll = jObject.getDouble("AHRSRoll");
                        AHRSGyroHeading = jObject.getDouble("AHRSGyroHeading");
                        AHRSMagHeading = jObject.getDouble("AHRSMagHeading");
                        AHRSSlipSkid = jObject.getDouble("AHRSSlipSkid"); // in degrees
                        AHRSTurnRate = jObject.getDouble("AHRSTurnRate");
                        AHRSGLoad = jObject.getDouble("AHRSGLoad");
                        AHRSGLoadMin = jObject.getDouble("AHRSGLoadMin");
                        AHRSGLoadMax = jObject.getDouble("AHRSGLoadMax");

                        GPSLatitude = jObject.getDouble("GPSLatitude");
                        GPSLongitude = jObject.getDouble("GPSLongitude");
                        GPSAltitudeMSL = jObject.getDouble("GPSAltitudeMSL");
                        GPSTrueCourse = jObject.getDouble("GPSTrueCourse");
                        GPSTurnRate = jObject.getDouble("GPSTurnRate");
                        GPSGroundSpeed = jObject.getDouble("GPSGroundSpeed");

                        BaroTemperature = jObject.getDouble("BaroTemperature");
                        BaroPressureAltitude = jObject.getDouble("BaroPressureAltitude");
                        BaroVerticalSpeed = jObject.getDouble("BaroVerticalSpeed");

                        // Status
                        //String status = getDeviceStatus();
                        //jObject = new JSONObject(status);

                        ///*
                        Log.v("b2", //" AHRSSlipSkid " + AHRSSlipSkid  +
                                    //" AHRSGLoad " + AHRSGLoad  +
                                    " AHRSTurnRate " + AHRSTurnRate +
                                    //" BaroTemperature " + BaroTemperature +
                                    " GPSTurnRate " + GPSTurnRate
                        );
                        // */

                    }
                    catch (JSONException e) {
                        e.printStackTrace();
                    }
                }
                finally {
                    mutex.release();
                }
            }
            catch (InterruptedException e) {
                e.printStackTrace();
            }
        }
`

ninelima, 2018-09-05:
`.779 23640-23673/player.efis.pfd V/b2:  AHRSTurnRate 3276.7 GPSTurnRate 0.0
`

ninelima, 2018-09-07:
Does anyone else experience the same problem?   
I am wondering if it might be my hardware.

ninelima, 2018-09-21:
I doubt it is hardware since I tested with another unit and the results are the same.
It also shows "- min" on the web app.

cyoung, 2018-09-21:
This is the value `ahrs.Invalid`. Likely because you're testing on the ground and not turning :)

https://github.com/cyoung/goflying/blob/0ba3e51be74b4848488e06ec0fee74ebaa7ef705/ahrs/ahrs_defs.go#L17

closed 2018-09-21 by cyoung

ninelima, 2018-10-16:
This definitely seems to be an issue.
I have not been able to anything other than 
	Invalid float64 = 3276.7     // 2**15-1

for the variable State.turnRate.