cyoung/stratux #744: Error during MAKE: No names found, cannot describe anything

jbrepogmailcom, 2018-09-05:
1. Stratux version:
1.4r5 downoaded from web as image

Hello, I have flashed image to SD card, connected over ssh and wanted to proceed by compilation guide (for Raspberry Zero W). But there was no /root/stratux directory. So I used git clone to get files (got a certificate error about github, but resolved that). Then I entered stratux directory and wanted to use make. Got these errors:

root@raspberrypi:~/stratux# make
make xdump978 xdump1090 xgen_gdl90 fancontrol
make[1]: Entering directory '/root/stratux'
cd dump978 && make lib
make[2]: Entering directory '/root/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o 
make[2]: Leaving directory '/root/stratux/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
git submodule update --init
Submodule 'dump1090' (https://github.com/stratux/dump1090) registered for path 'dump1090'
Submodule 'goflying' (https://github.com/cyoung/goflying) registered for path 'goflying'
Cloning into 'dump1090'...
remote: Counting objects: 4120, done.
remote: Total 4120 (delta 0), reused 0 (delta 0), pack-reused 4120
Receiving objects: 100% (4120/4120), 5.57 MiB | 1.30 MiB/s, done.
Resolving deltas: 100% (2810/2810), done.
Checking connectivity... done.
Submodule path 'dump1090': checked out '9a4fb850937565cfeadd1e5889cddbf93f45faf5'
Cloning into 'goflying'...
remote: Counting objects: 1482, done.
remote: Total 1482 (delta 0), reused 0 (delta 0), pack-reused 1481
Receiving objects: 100% (1482/1482), 437.40 KiB | 463.00 KiB/s, done.
Resolving deltas: 100% (866/866), done.
Checking connectivity... done.
Submodule path 'goflying': checked out '0ba3e51be74b4848488e06ec0fee74ebaa7ef705'
cd dump1090 && make
make[2]: Entering directory '/root/stratux/dump1090'
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W `pkg-config --cflags librtlsdr`  -c dump1090.c -o dump1090.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c anet.c -o anet.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c interactive.c -o interactive.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c mode_ac.c -o mode_ac.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c mode_s.c -o mode_s.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c net_io.c -o net_io.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c crc.c -o crc.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c demod_2000.c -o demod_2000.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c demod_2400.c -o demod_2400.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c stats.c -o stats.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c cpr.c -o cpr.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c icao_filter.c -o icao_filter.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c track.c -o track.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c util.c -o util.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c convert.c -o convert.o
gcc -g -o dump1090 dump1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o crc.o demod_2000.o demod_2400.o stats.o cpr.o icao_filter.o track.o util.o convert.o -lpthread -lm -lrt `pkg-config --libs librtlsdr` 
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c view1090.c -o view1090.o
gcc -g -o view1090 view1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o crc.o stats.o cpr.o icao_filter.o track.o util.o -lpthread -lm -lrt 
make[2]: Leaving directory '/root/stratux/dump1090'
go get -t -d -v ./main ./godump978 ./uatparse ./sensors
make[1]: go: Command not found
Makefile:17: recipe for target 'xgen_gdl90' failed
make[1]: *** [xgen_gdl90] Error 127
make[1]: Leaving directory '/root/stratux'
Makefile:14: recipe for target 'all' failed
make: *** [all] Error 2
root@raspberrypi:~/stratux# make install
cp -f gen_gdl90 /usr/bin/gen_gdl90
cp: cannot stat ‘gen_gdl90’: No such file or directory
Makefile:40: recipe for target 'install' failed
make: *** [install] Error 1
root@raspberrypi:

jbrepogmailcom, 2018-09-05:
Do not understand why after pasting the terminal output text is overcrossed. It should not be

cyoung, 2018-09-05:
Remove the `/root/stratux` directory and clone it once again using the command `cd /root ; git clone --recursive https://github.com/cyoung/stratux`.

closed 2018-09-05 by cyoung

jbrepogmailcom, 2018-09-05:
Hello.

It downloaded some more files but ended up with same error.

Jan

aruly, 2018-09-05:
Sent from my BlackBerry 10 smartphone.                                                                                                                                                                                                                From: cyoungSent: Wednesday, September 5, 2018 3:41 PMTo: cyoung/stratuxReply To: cyoung/stratuxCc: SubscribedSubject: Re: [cyoung/stratux] Error during MAKE: No names found, cannot describe anything (#744)Remove the /root/stratux directory and clone it once again using the command cd /root ; git clone --recursive https://github.com/cyoung/stratux.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/cyoung/stratux/issues/744#issuecomment-418874083",
"url": "https://github.com/cyoung/stratux/issues/744#issuecomment-418874083",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [cyoung/stratux] Error during MAKE: No names found, cannot describe anything (#744)",
"sections": [
{
"text": "",
"activityTitle": "**cyoung**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@cyoung",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"cyoung/stratux\",\n\"issueId\": 744,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/cyoung/stratux/issues/744#issuecomment-418874083"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 376338583\n}"
}
],
"themeColor": "26292E"
}
], ‎

pvanessen, 2019-05-12:
Have the same issue. 
`root@routerpi:~# rm -rf stratux/
root@routerpi:~# git clone --recursive https://github.com/cyoung/stratux
Cloning into 'stratux'...
remote: Enumerating objects: 9492, done.
remote: Total 9492 (delta 0), reused 0 (delta 0), pack-reused 9492
Receiving objects: 100% (9492/9492), 43.26 MiB | 2.35 MiB/s, done.
Resolving deltas: 100% (6462/6462), done.
Checking connectivity... done.
Submodule 'dump1090' (https://github.com/stratux/dump1090) registered for path 'dump1090'
Submodule 'goflying' (https://github.com/cyoung/goflying) registered for path 'goflying'
Cloning into 'dump1090'...
remote: Enumerating objects: 4120, done.
remote: Total 4120 (delta 0), reused 0 (delta 0), pack-reused 4120
Receiving objects: 100% (4120/4120), 5.57 MiB | 2.01 MiB/s, done.
Resolving deltas: 100% (2810/2810), done.
Checking connectivity... done.
Submodule path 'dump1090': checked out '9a4fb850937565cfeadd1e5889cddbf93f45faf5'
Cloning into 'goflying'...
remote: Enumerating objects: 1487, done.
remote: Total 1487 (delta 0), reused 0 (delta 0), pack-reused 1487
Receiving objects: 100% (1487/1487), 442.65 KiB | 0 bytes/s, done.
Resolving deltas: 100% (869/869), done.
Checking connectivity... done.
Submodule path 'goflying': checked out '0ba3e51be74b4848488e06ec0fee74ebaa7ef705'
root@routerpi:~# cd stratux/
root@routerpi:~/stratux# make all
make xdump978 xdump1090 xgen_gdl90 fancontrol
make[1]: Entering directory '/root/stratux'
cd dump978 && make lib
make[2]: Entering directory '/root/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o 
make[2]: Leaving directory '/root/stratux/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
sudo: unable to resolve host routerpi
git submodule update --init
cd dump1090 && make
make[2]: Entering directory '/root/stratux/dump1090'
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W `pkg-config --cflags librtlsdr`  -c dump1090.c -o dump1090.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c anet.c -o anet.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c interactive.c -o interactive.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c mode_ac.c -o mode_ac.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c mode_s.c -o mode_s.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c net_io.c -o net_io.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c crc.c -o crc.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c demod_2000.c -o demod_2000.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c demod_2400.c -o demod_2400.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c stats.c -o stats.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c cpr.c -o cpr.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c icao_filter.c -o icao_filter.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c track.c -o track.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c util.c -o util.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c convert.c -o convert.o
gcc -g -o dump1090 dump1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o crc.o demod_2000.o demod_2400.o stats.o cpr.o icao_filter.o track.o util.o convert.o -lpthread -lm -lrt `pkg-config --libs librtlsdr` 
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c view1090.c -o view1090.o
gcc -g -o view1090 view1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o crc.o stats.o cpr.o icao_filter.o track.o util.o -lpthread -lm -lrt 
make[2]: Leaving directory '/root/stratux/dump1090'
go get -t -d -v ./main ./godump978 ./uatparse ./sensors
package context: unrecognized import path "context" (import path does not begin with hostname)
package net/http/httptrace: unrecognized import path "net/http/httptrace" (import path does not begin with hostname)
Makefile:17: recipe for target 'xgen_gdl90' failed
make[1]: *** [xgen_gdl90] Error 1
make[1]: Leaving directory '/root/stratux'
Makefile:14: recipe for target 'all' failed
make: *** [all] Error 2`