cyoung/stratux #746: Stratux not picking up towers

joeysweet, 2018-09-06:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ x] dual

    GPS
      - [ ] yes
      - [ x] no
      type:

    AHRS
      - [ ] yes
      - [x ] no

    power source: Aircraft Power

    usb cable: Generic micro USB

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
Foreflight v9.6
    EFB platform: (e.g., iOS 9.2)
ios 7.01.000
    EFB hardware: (e.g., iPad Mini 2)
Ipad mini 2
4. Description of your issue:
I have used the stratux at low altitude recently and had somewhat poor reception, but still would pick towers up.

Up in the flight levels I can't seem to find any towers out of the side window in a Pilatus PC-12. The stratuS works well and can find antennas even in the side bin - but the stratux with high gain antennas in the window will not pick up any up. I have tried rebooting in the air and making sure the 978/1090 options are turned on, as well as updating the firmware of the stratus to the latest version.

Thanks in advance y'all!
If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

joeysweet, 2018-09-06:
Correction- updating the firmware on the stratuX!

Helno, 2018-09-08:
Might just be a matter of placing it near a different window. Often the heated or metalized windows cause reception issues. 

How many towers are you typically seeing when down low?

Up it the flight levels it gets harder to get good reception but it is still possible.

nikitaxik, 2018-09-19:
I have also observed degraded reception performance in the last release or two.  I often fly in SFRA around Washington DC and I used to get at least 5-6 towers at 5000 ft all the time.  Now I get 1 or maybe 2 and sometimes none.  I almost always have a message on my ForeFlight about marginal reception.
I cannot really identify when this degradation showed up exactly, but it was around the time ForeFlight implemented support for AHRS and I upgraded to the Stratux software release that was ForeFlight compatible.
I have other pilots report degraded reception performance as well.  I do not think it is just the placement of the unit.  I am suspecting that something changed in the code of the last two releases which caused this degradation.  Are there any performance tests that can be run to see if there is an issue?  I strongly suspects there is.

cyoung, 2018-09-21:
@nikitaxik - it sounds like you're getting the same reception as before, but ForeFlight has now chosen to call it "marginal" when they previously called it "1 tower" or "2 tower". They've just changed the way they report it to you.

cyoung, 2018-10-10:
@joeysweet - We have a new 978 radio that is more sensitive, send me a note via email if you want to give it a try.

srhalter, 2018-11-05:
In have also noticed towers are hard to pick up at 6000 ft, I am running the current release. I also did a reboot at 6000 ft and then quickly picked up 2 towers, but then 1 went away.

cyoung, 2018-11-05:
@joeysweet - try the uatradio.

What area are you in @srhalter? Also might try the new 978 radio.

srhalter, 2018-11-05:
I'm on Central Wisconsin, I usually fly at 3000msl and normally get one
tower where ever I am, just happened to notice I was at 6000 msl and wasn't
seeing any for a while,

closed 2018-11-26 by cyoung