cyoung/stratux #752: Use Versioning for Golang

doodles526, 2018-10-11:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

We shouldn't want external package changes to break our builds. We should be using Golang versioning. We can change to `vgo` when officially supported by the Go 1.12 release. But for now can use `glide` or similar.

Also best-practice for imports should include full path names. Eg: https://github.com/cyoung/stratux/blob/master/main/sensors.go#L11 should be changed to be `github.com/cyoung/stratux/goflying/ahrs` -> or another way would be to version the goflying repo against it using golang version controls. 

This should likely fix https://github.com/cyoung/stratux/issues/723 as well, as it will resolve the non-local import issues by having all nescessary dependencies in the `./vendor` dir

cyoung, 2018-11-26:
Agreed. We're using git submodules to achieve this currently. Should we wait for `vgo`, do you think?

doodles526, 2018-11-27:
@cyoung I think that's a good idea. Looking at this issue, it looks like module support will be fully integrated with 1.12, slated to be released in ~2 months https://github.com/golang/go/issues/27563

doodles526, 2020-03-19:
WIP https://github.com/cyoung/stratux/pull/809