cyoung/stratux #757: Stratux GPS accuracy is very poor.

capilee315, 2018-10-26:
1. Stratux version: v1.4r5 (991f48eaa5)
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type: GPYes 2.0 u-blox 8 GPS unit

    AHRS
      - [ X] yes
      - [ ] no

    power source: External Battery Pack Pocket Juice Endurance 3X

    usb cable:

3. EFB app and version: Foreflight 10.4.3

    EFB platform: iOS 12.0.1

    EFB hardware: iPad Mini (Lastest Version)

4. Description of your issue:

Greetings!
I recently purchased Stratux.
I tried to get the GPS to work.
However, the GPS accuracy in Foreflight shows 93 Meters.
I thought it could be the image issue, so I re-imaged the SD card and still it shows poor GPS accuracy.
Should I purchase new GPS chip or should I re-image the SD card and try it again?
Thank you for your help.

capilee315, 2018-10-26:
> Do you have this in a plastic case? Are you testing in your aircraft or another location? A friend had another brand of GPS chip in a metal enclosure and had bad luck.
> 
> I use the VK-162 external GPS dongle and almost always get 10m or less accuracy if I give it a good view of the sky.

Thanks for the reply. I have it in a plastic case. Only thing that is between the GPS reciever is the plastic cover. I'm testing it outside of my house. Do you think, should I change it to VK-162 external?

ninelima, 2018-10-26:
I had a similar problems on 2 occasions.

The GPS was not properly plugged into the USB port. I think it had 
vibrated out or the unit got bumped. It was as simple as that;

The second time was dud/inetrmittent GPS receiver

capilee315, 2018-10-26:
> I had a similar problems on 2 occasions. The GPS was not properly plugged into the USB port. I think it had vibrated out or the unit got bumped. It was as simple as that; The second time was [...]

I disarmed the Stratux and tried to see what could interfere with the GPS antennae..
Still nothing.. I wonder if is just my GPS reciever is faulty... Should I just ask to Stratux guys about the issue?

lalalandrus, 2018-11-26:
The pi and the SDR radios are extremely noisy and easily overpowers the weak GPS signal. My reception was extremely poor until I unplugged the SDR receivers or extended the GPS far away from the unit. I suggest looking on websites like thingiverse to see what others have implemented for their shielding. 

I ended up using a metal case for my shielding https://www.reddit.com/r/stratux/comments/9ubvi7/pi_zero_w_in_metal_case/

cyoung, 2018-11-26:
Newer UAT radio is less noisy and helps out a bit.

The receiver does sound bad, send a note to the vendor @capilee315.

closed 2018-11-26 by cyoung