cyoung/stratux #759: Intermittent WebSocket Drop Off

JohnMarzulli, 2018-11-13:
1. Stratux version:
1.4r5
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source:
        _*5v USB power socket AND Anker PowerCore 26800*_

    usb cable:
        _*Anker*_

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
        _*StratuxHud, 1.3*_
    EFB platform: (e.g., iOS 9.2) 
        _*Stratux 1.4r5 (Custom Image)*_
    EFB hardware: (e.g., iPad Mini 2)
        *_Raspberry Pi 3_*

4. Description of your issue:
The StratuxHud implements an ICMP Ping protocol. This sends Ping/Pong requests/responses to the Stratux.

When the software is running on the Stratux unit, after about 10 minutes messages from the WebSocket are no longer received. When the ping is coming from a dedicated unit (separate Raspberry Pi), the websocket stops receiving data after about 20 minutes.

Inspection of the Stratux code shows that receiving a ICMP RESPONSE should keep the client out of sleep mode. The HUD is making sure that a ICMP_ECHO_REPY is being sent, and sends one every two seconds "just to be sure".

Inspection of the Stratux code (I'm not fluent in Go...) looks like a timeout/sleep should happen within 15 seconds.

Additionally, I can not reproduce this behavior "on the ground".

cyoung, 2018-11-26:
John, I think this is an issue in the websocket library. Can you try writing to the websocket every so often to keep it alive? The content you send doesn't matter, it will be discarded. If that keeps the socket alive indefinitely, it's the same issue as another app integrator had at one time.

cyoung, 2018-11-27:
>To be clear, you are suggesting writing to the same socket that is already open and I am receiving the traffic updates on?

Yes, exactly.