cyoung/stratux #761: Stratux Stuck "Connecting" in Windows

lalalandrus, 2018-11-26:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type:

    AHRS
      - [X ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
    FLTPlanGo Latest
    EFB platform: (e.g., iOS 9.2)
    Windows 
    EFB hardware: (e.g., iPad Mini 2)
    Windows 10 tablet
4. Description of your issue:

Stratux typically has no gateway served from the DHCP as it has no internet connection which is corrent. However in windows, this causes the network to be classified as "Unidentified" and can only be classified as a "Public" networks. 

Default settings for the windows firewall for public network is to not respond to ICMP pings. Therein lies the issue. Confirmed via wireshark and the stratux code, if the device does not answer to pings, no data is sent. This causes FLTPlanGo to be stuck in "Connecting" which is correct. 

While I dont have a good solution, perhaps there could be a way to remove the wait for ping requirement before sending the data. 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2018-11-26:
>Stratux typically has no gateway served from the DHCP

If that's the case, you may be running a really old version or something else is wrong. It should be serving 192.168.10.1 as the default gateway.

https://github.com/cyoung/stratux/blob/master/image/dhcpd.conf#L113

lalalandrus, 2018-11-26:
i am using a version downloaded a month ago from github and compiled locally as i am using a RPIZeroW.

i did add the "options router" but it crashed DHCPd. ill import that conf file later tonight to see if it helps.

cyoung, 2018-11-26:
"options router" is included by default. If it crashes dhcpcd then there is probably something else wrong...

lalalandrus, 2018-11-26:
i just had a check, windows firewall will block all ping requests - private or public. i dont think the gateway will will change anything but i will try with the new conf file and see if it will work without the work around

cyoung, 2018-11-26:
A workaround that would work for everyone would be ideal. Can we easily detect that the client is a Windows machine?

One-off fix would be to remove these lines and recompile:
https://github.com/cyoung/stratux/blob/master/main/network.go#L150-L152

lalalandrus, 2018-11-27:
i tested it and it worked, 

i return false on all conditions in that function
https://github.com/cyoung/stratux/blob/master/main/network.go#L150-L152

also since my pi zero takes over half and hour to compile, i also removed this line 
https://github.com/cyoung/stratux/blob/e5916be47226e385bcc892a2d78eb5208d882a26/main/network.go#L532

as mentioned before, maybe this ping requirement can be removed without introducing another issue.

cyoung, 2018-12-04:
It seems like it could be removed in your case.

The idea with sleep mode functions is to buffer weather data while a tablet is not able to accept data (app not in foreground, screen turned off).

See [here](https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.md#sleep-mode).

closed 2018-12-04 by cyoung