cyoung/stratux #762: Default Image DHCP options

Michandrz, 2018-11-30:
1. Stratux version: 1.4r5
2. Stratux config: 

    SDR
      - [ ] single
      - [ X ] dual

    GPS
      - [ x ] yes
      - [ ] no
      type:

    AHRS
      - [ x ] yes
      - [ ] no

    power source: Battery bank

    usb cable: yes

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Foreflight 

    EFB platform: (e.g., iOS 9.2) IOS 11.1.2

    EFB hardware: (e.g., iPad Mini 2) Iphone 6s 64gb

4. Description of your issue:

more of a suggestion for the image than an issue:

/etc/httpd/httpd.conf comment out the line reading default router set to 192.168.10.1

By doing this, you can still connect to stratux, however it signals to IOS devices that this connection does not provide internet access and allows them to use cellular data for non-status communications.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

wa9als, 2018-11-30:
I just make the connection a static connection instead of automatic, and all the other stuff can use cellular.

cyoung, 2018-12-04:
Causes issues with Android and other platforms when there is no default route.

We had it configured this way for a while for cellular access, but the feature wasn't worth the trouble.

Stratux is for use in flight. You won't have any decent cellular coverage.

closed 2018-12-04 by cyoung