cyoung/stratux #764: rtl-sdr.h   missing....

ppsel03, 2018-12-04:
1. Stratux version:V1.5b1
2. Stratux config: 
    power source:LiPo
    Raspberry Pi 3b+  "Raspbian Lite"

3. EFB app and version: (ForeFlight, Avare)
    EFB platform: (Android, iOS)
    EFB hardware: (Nexus 5, iPad )

4. Description of your issue:

  Hi, love the product.  Does V1.5b1 contain all the dev tools?   
  
From "Raspbian Lite" Followed  [How to compile and build Stratux Executable](https://github.com/cyoung/stratux/wiki/How-to-compile-and-build-Stratux-executable#explanation)

    sudo su -
    cd /root/
    wget https://dl.google.com/go/go1.10.4.linux-armv6l.tar.gz
    tar -zxvf go1.10.4.linux-armv6l.tar.gz
    ...sudo apt-get install git
    git clone --recursive https://github.com/cyoung/stratux
    cd stratux
    service stratux stop
    ...GOROOT=/root/go
    ...PATH=$PATH:$GOROOT/bin
 >>>   make
    make install
    service stratux start 

make throws No package 'librtlsdr' found.

output:   

   root@raspberrypi:~/stratux# make
   make xdump978 xdump1090 xgen_gdl90 fancontrol
   make[1]: Entering directory '/root/stratux'
   cd dump978 && make lib
   make[2]: Entering directory '/root/stratux/dump978'
   gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c 
   fec/init_rs_char.c
   gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o
   make[2]: Leaving directory '/root/stratux/dump978'
   sudo cp -f ./libdump978.so /usr/lib/libdump978.so
   git submodule update --init
   cd dump1090 && make
   make[2]: Entering directory '/root/stratux/dump1090'
   fatal: No names found, cannot describe anything.
   gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W `pkg-config --cflags librtlsdr`  -c 
   dump1090.c -o dump1090.o
   Package librtlsdr was not found in the pkg-config search path.
   Perhaps you should add the directory containing `librtlsdr.pc'
   to the PKG_CONFIG_PATH environment variable
   No package 'librtlsdr' found
   dump1090.c:52:21: fatal error: rtl-sdr.h: No such file or directory
     #include <rtl-sdr.h>
                     ^
   compilation terminated.
   Makefile:37: recipe for target 'dump1090.o' failed
   make[2]: *** [dump1090.o] Error 1
   make[2]: Leaving directory '/root/stratux/dump1090'
   Makefile:25: recipe for target 'xdump1090' failed
   make[1]: *** [xdump1090] Error 2
   make[1]: Leaving directory '/root/stratux'
   Makefile:14: recipe for target 'all' failed
   make: *** [all] Error 2

ppsel03, 2018-12-04:
Added before make :

   apt-get install librtlsdr0 librtlsdr-dev 
   export GOPATH=/root/go_path

But, make threw : 

root@MyStratux:~/stratux# make
make xdump978 xdump1090 xgen_gdl90 fancontrol
make[1]: Entering directory '/root/stratux'
cd dump978 && make lib
make[2]: Entering directory '/root/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o
make[2]: Leaving directory '/root/stratux/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
git submodule update --init
cd dump1090 && make
make[2]: Entering directory '/root/stratux/dump1090'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/stratux/dump1090'
go get -t -d -v ./main ./godump978 ./uatparse ./sensors
go build -ldflags "-X main.stratuxVersion=`git describe --tags --abbrev=0` -X main.stratuxBuild=`git log -n 1 --pretty=%H`" -p 4 main/gen_gdl90.go main/traffic.go main/gps.go main/network.go main/managementinterface.go main/sdr.go main/ping.go main/uibroadcast.go main/monotonic.go main/datalog.go main/equations.go main/sensors.go main/cputemp.go main/lowpower_uat.go
go build _/root/stratux/godump978: invalid flag in #cgo CFLAGS: -L/root/stratux
# github.com/jpoirier/gortlsdr
../go_path/src/github.com/jpoirier/gortlsdr/rtlsdr.go:596:11: could not determine kind of name for C.rtlsdr_set_bias_tee
Makefile:17: recipe for target 'xgen_gdl90' failed
make[1]: *** [xgen_gdl90] Error 2
make[1]: Leaving directory '/root/stratux'
Makefile:14: recipe for target 'all' failed
make: *** [all] Error 2

cyoung, 2018-12-04:
See here for some clues. v1.5b1 image doesn't contain all of the required sources.

https://github.com/cyoung/stratux/blob/master/image/mkimg.sh#L104-L107

ppsel03, 2018-12-04:
Found 3B+ supported image here.  https://github.com/cobraPA/stratuxMorePi/releases  so working with it.

cyoung, 2018-12-04:
You'll have to modify the paths some.

closed 2018-12-04 by cyoung

ppsel03, 2018-12-04:
Unzipped and wrote https://github.com/cobraPA/stratuxMorePi/releases/tag/v1.5b1%2B  to SDCard and came STRATUX came up on the Pi 3B+ and showed airplane traffic immediately (AVARE) .