cyoung/stratux #766: Stratux Disconnect during flight

NSalai, 2019-02-17:
1. Stratux version: v1.4r5
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [ ] yes
      - [x] no
      type:

    AHRS
      - [x] yes
      - [ ] no

    power source: EasyAcc Powerbank PB6000CB w/built-in USB-micro

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) ForeFlight 11.0.1

    EFB platform: (e.g., iOS 9.2) iOS 12.1.4

    EFB hardware: (e.g., iPad Mini 2) 2017 iPadPro 10.5"

4. Description of your issue:

So apologies for my naivete, I'm a flight instructor who uses the Stratux build paired with ForeFlight. As such I use it quite a bit (anytime I leave the traffic pattern), recently I have encountered an issue where I get an aural warning that "Stratux has disconnected" and visually all traffic has been removed from ForeFlight. If I click on settings, the unit still shows connected and typically within 20-40 seconds all traffic is repopulated and the system works as intended. 

Over the past 36 hours and the course of 3 flights, I had numerous disconnects some mere seconds apart. Any help would be greatly greatly appreciated. 

Attached is the device log and master log from ForeFlight.

> 145 Stratux disconnected. Reason: No longer receiving messages.

>203 PANEL_CONNECTION_STATE_CHANGED: Stratux disconnected. Reason: No longer receiving messages.
> 203 PANEL_CONNECTION_STATE_CHANGED: No devices connected.
> 203 PANEL_CONNECTION_STATE_CHANGED: Stratux connected. Reason: Device just connected.
> 203 PANEL_CONNECTION_STATE_CHANGED: v1.4r5-991f48eaa is connected

[DeviceDiagnostic-Nick_s_iPad-7Days (1).zip](https://github.com/cyoung/stratux/files/2872499/DeviceDiagnostic-Nick_s_iPad-7Days.1.zip)

-Nick

cyoung, 2019-02-24:
Hi Nick, could you update to [v1.5b2](https://github.com/cyoung/stratux/releases/tag/v1.5b2) and see how it does?

NSalai, 2019-02-25:
Chris,

Updated to v1.5b2 this past Friday, two flights on Saturday, no disconnects. I did start to shutdown the unit from the Stratux menu, before I was unplugging the power before. Don't know if it was causing corruption??

Installed the new v3 UAT antenna yesterday, another flight tomorrow and Tuesday, both several hours long so I'll update with how it does. 

Nick

NSalai, 2019-02-27:
> Hi Nick, could you update to [v1.5b2](https://github.com/cyoung/stratux/releases/tag/v1.5b2) and see how it does?

Two flights, and still have an issue. Had a singular disconnect on a 2.0 flight today. No disconnects last night on 2.5 hour flight. 

Thoughts?

davecessna, 2019-03-29:
I am also an instructor flying out of the NYC area. I notice when very close to the hub airports, the saturation of traffic causes my Stratux to lock up and disconnect. I don't receive a message in my EFB however, WIFI stays connected to the device but I can't access the webgui. To solve the issue, I can disable the 1090 radio when flying around there and just rely on the 978 for the TIS-B traffic relays.

NSalai, 2019-07-16:
So this is **STILL** occurring, random disconnects sometimes hours apart, other times mere seconds apart. All temperature ranges, all time ranges, and all location ranges. Do we have any idea as as to what this might be? Has there been any new software updates?

Helno, 2019-07-28:
This looks like it might be a recurrence of #631 

I saw a few random drop outs on Foreflight during my trip to and from Oshkosh. I saw about 3 or 4 drop outs on my Stratux and a few times on my Flightstream 210. They never happened at the same time and neither required an app shutdown or Stratux reboot to fix.

I'll send them an email to see if they are aware of the issue.

NSalai, 2019-07-28:
Good to know, I do not lose WiFi connection to the Stratux unit and the traffic is repopulated on the screen within 5-10 seconds. Very odd. 

I have talked to a few other instructors where I teach and the ones using Stratux's (6 of us) and ForeFlight's Scout (2) report random dropouts/disconnects, most of them reporting that it's to the west of KDVT within 4-10 miles... We are not sure if there is someone with high powered antenna's on the ground or if it's the nearby military base (KLUF)

Austinthemighty, 2019-08-25:
@Nsalai On my way back from KHOU to KPHX, ForeFlight kept reporting that it had lost its connection from my stratux device. I don’t think it’s KLUF. Even on my way to. KHOU it kept dropping, so I don’t know.  It might be a bad solder joint on the WiFi module or something causing the PCB to flex and a trace breaking, might be heat if the PCBs flexing.

NSalai, 2019-08-25:
As Helno explained to me, it appears to be something in ForeFlight that is causing the disconnects...

Used a students Sentry this morning while banging around the 'stack' and then shot a few approaches at SDL and DVT, no issues whatsoever.

NSalai, 2019-08-25:
@Austinthemighty As Helno explained to me, it appears to be something in ForeFlight that is causing the disconnects...

Used a students Sentry this morning while banging around the 'stack' and then shot a few approaches at SDL and DVT, no issues whatsoever.

davecessna, 2019-08-30:
I don’t think it’s foreflight. I’m having the issue in fltplan go

Austinthemighty, 2019-08-30:
It could also be an issue with iOS.

nzkarit, 2019-09-15:
Hi,

This may be a dump1090 issue. I am currently doing some investigation into ADS-B handling and looking at issues. (Currently broadcasting on ISM frequencies and have Dump1090 to listen to ISM frequency (ISM is public non aviation frequency), using https://github.com/nzkarit/ADSB-Out, which is how I can correct the number of aircraft etc)

For instance if there are 1200-1300 aircraft received by dump1090, the list of aircraft sent to the web browser view (http://172.24.51.115:8080/dump1090/data.json), the JSON response is just truncated, in that it stops mid aircraft entry:
```
<snip>
{"hex":"003724", "squawk":"0000", "flight":"CC003724", "lat":12.340021, "lon":56.779994, "validposition":1, "altitude":9875,  "vert_rate":14784,"track":19, "validtrack":1,"speed":447, "messages":6, "seen":6},
{"hex":"003723", "squawk":"0000", "flight":"CC003723", "lat":12.340021, "lon":56.779994, "validposition":1, "altitude":9875,  "vert_rate":14784,"track":19, "validtrack":1,"speed":447, "messages":6, "seen":6},
{"hex":"003722", "squawk":"0000", "flight":"CC003722", "lat":12.340021, "lon":56.779994, "validposition":1, "altitude":9875,  "vert_rate":14784,"track":19, "vali
```

I have only thus far looked at the dump1090 webui and not at the other interfaces which dump1090 exposes, so I currently do not know if they get truncated as well. I have not fully researched the issue and thus have not submitted the issue to dump1090 yet.

For Stratux it may be worth ensuring that the ADS-B ingestion code is checked to see how it handles partial records and truncated data. Likewise the EFB application may not have the correct error handling in place for this situation and it showing a disconnected status when it receives partial records.

Hope that makes sense?

Or this may be totally unrelated.

Helno, 2019-12-16:
Anyone still seeing random disconnects? I haven't seen any from Stratux in a while but saw one with a GTX345 this weekend.

kiltedpiper, 2019-12-16:
I am having random disconnects.  Happens throughout a flight.  I have not been able to isolate it to certain areas or circumstances.  I am using ForeFlight.

davecessna, 2019-12-16:
I’m fortunate enough to have moved to the perspective plus setup which I believe has the flight stream 510. I have not had any disconnects with foreflight since I started working with Cirrus.

NSalai, 2019-12-16:
> Anyone still seeing random disconnects? I haven't seen any from Stratux in a while but saw one with a GTX345 this weekend.

Yes, consistently inconsistent disconnects, primarily when around/over Bravo and near busy Delta airspace. 

Also seeing different data being populated as compared to some of our G650s and G1000s w/ traffic. Sometimes more and sometimes less.

d-c-d, 2020-09-21:
This happened to me multiple times yesterday - ( I'll need to update, and check configuration, etc )
I know it's been quite a while since my last software update, and for all I know it could have been an 'operator error'

NSalai, 2020-09-22:
Honestly, its been a constant issue with no resolution or update. So I've given up, I'm glad it works ok for the developers with their 2-3 flights a year in quiet airspace. But as someone who flys 4-5 times a day, sometimes up to 8 hours in and around a bravo and *literally* the busiest GA airport in the country (KDVT) I cant put up with this anymore. I'm moving on to Sentry, and so should you.

Austinthemighty, 2020-09-22:
> Honestly, its been a constant issue with no resolution or update. So I've given up, I'm glad it works ok for the developers with their 2-3 flights a year in quiet airspace. But as someone who flys [...]

Ive been flying out of KGEU (around 14nm from KDVT) twice a week for the last 3 weeks and I haven't had a problem, maybe it's a hardware issue and not a software issue, or maybe it's a compatible issue. I was using a iPad Air 1 and I was having disconnect issues when I flew into KPHX, since moving to a new iPad 6 gen I haven't had issues.

Austinthemighty, 2020-09-22:
What version pi are you using? Also try changing the positions of the SDRs, and see if that makes a difference, it could also be that the battery bank is not supplying enough voltage or amps, and the pi then reboots. Or a PCB is damaged and when it heats up a trace expands and disconnects causing a short or no power.

dbarrington1, 2020-09-22:
I think you underestimate the number of people who use this... it’s enough that even Foreflight decided to support it without a basic hack...

It works great for me and I fly a lot... I’m not a developer and know proactively nothing about programing...
I did make sure to power it with an Anker power supply plugged into ships power, and run an external antenna for the lower frequency and the rubber ducky for the higher frequency...

Works amazing for me

NSalai, 2020-09-22:
> What version pi are you using? Also try changing the positions of the SDRs, and see if that makes a difference, it could also be that the battery bank is not supplying enough voltage or amps, and [...]

NSalai, 2020-09-22:
Im using the *exact* parts list from the Stratux website, assembled per the video running the latest build. I am using an iPad Pro. I've even tried just plugging it straight into the wall, its not a power supply issue. As for a short in the PCB, that would require a complete system restart, something that does not happen. It will repopulate within 5-20 seconds.

cyoung, 2020-09-22:
@NSalai - what version of ForeFlight are you running?

NSalai, 2020-09-22:
> I think you underestimate the number of people who use this... it’s enough that even Foreflight decided to support it without a basic hack... It works great for me and I fly a lot... I’m not a [...]

There *were* 6 of us instructors using it at our flight school, roughly 8000ish flight hours logged between us in the past 2 years, but *we* are the outliers?? We have all had issues, per my earlier comments, since very early on. Again, I bought *everything* right from the Stratux webpage links off of Amazon, assembled *exactly* using the guide. Two of the instructors bought pre-assembled units. 

There is something in this airspace that is causing data loss whether it be traffic saturation, data dumps, or signal jamming. I can predict almost down to the exact spot where/when I lose Stratux and all traffic when I fly over KPHX actual.

NSalai, 2020-09-22:
> @NSalai - what version of ForeFlight are you running?

12.7.2

cyoung, 2020-09-22:
If you're willing to experiment, there are a few things that could be tried. Changing the WiFi channel might be a start. "No longer receiving messages" seems to indicate that ForeFlight isn't even receiving heartbeat messages, so maybe there's some problem with the WiFi connection.

Did you test the other device (Sentry) with the same iPad in the same aircraft in the same area?

NSalai, 2020-09-22:
> If you're willing to experiment, there are a few things that could be tried. Changing the WiFi channel might be a start. "No longer receiving messages" seems to indicate that ForeFlight isn't even [...]

Yes. I've tried Sentry and Scout. Both work with no issues. 

If you want to try some new settings I'm willing.

cyoung, 2020-09-22:
I'd start with updating to [v1.6r1](https://github.com/cyoung/stratux/releases/tag/v1.6r1) and using the new WiFi settings on the interface (http://192.168.10.1/, "Settings"). The default WiFi channel is 1. You can try running it on channel 6 or channel 11. Since you fly often, making a single change and testing it out might be the best thing. I appreciate your patience. You'd be surprised about the issues that even certified equipment has working together sometimes.

rhole, 2020-09-22:
Could this be an antenna issue?  Perhaps placement is giving a dead zone.
Rick Hole

cyoung, 2020-09-22:
@rhole - It doesn't seem to be an ADS-B reception issue. It seems like a problem with ForeFlight and Stratux maintaining connectivity. That is, if ForeFlight is still throwing the same disconnect messages as in the original issue description.

NSalai, 2020-09-22:
> I'd start with updating to [v1.6r1](https://github.com/cyoung/stratux/releases/tag/v1.6r1) and using the new WiFi settings on the interface (http://192.168.10.1/, "Settings"). The default WiFi [...]

Ok. I'm walking out the door now, and have prog checks/flights all day so I wont be able to update and change settings until tonight/tomorrow at the soonest. 

I will keep you informed. Thanks for replying, although I do wish we could have had this dialog sooner. 😖

NSalai, 2020-09-22:
> @rhole - It doesn't seem to be an ADS-B reception issue. It seems like a problem with ForeFlight and Stratux maintaining connectivity. That is, if ForeFlight is still throwing the same disconnect messages as in the original issue description.

I fly 172s and pa44s, I've tried putting the stratux in every conceivable place and orientation over the past 18 months... Does not seem to matter.

saabnut, 2020-09-23:
Nice... can you provide th coordinates of that predictable location?
In the early 80's ford had a number of cars that would stall intermittently
due to ground based RFI. A special shielded ecu was the fix. If the emitter
in your case is messing with 978, or 1090 data, the FCC can fix it. If its
a 2.4ghz wifi signal, maybe not.

saabnut, 2020-09-23:
Does it happen with a different software product on the same Ipad? Get the
trial from Seattle avionix, and fly the same location. That should identify
a network interference issue.

Austinthemighty, 2020-09-23:
I would be interested in seeing the flight logs and everywhere it disconnects

NSalai, 2020-09-23:
Without fail, over KPHX, everytime I do a bravo transition...

saabnut, 2020-09-23:
Could it be interference from shaw butte microwave relay station, about 8
NNW of KPHX?

aschi2, 2020-12-24:
Any update on this? I have this exact same issue.

mhrussell1, 2021-04-25:
I would also like an update on this issue. I have an external GPS and the new low power SDRs.  I have updated to 1.6r1 and I am still having traffic dropouts for 5+ minutes and when it comes back it does not stay.  The web interface indicates no connection to Stratux.  I tried switching to channel 7 on the Wifi.  I have a stratux logfile from yesterday.  I know I was seeing traffic at 2222 UTC but dropped out after that to the end of the flght at 2240.

[2021-04-24-stratux.log](https://github.com/cyoung/stratux/files/6372288/2021-04-24-stratux.log.txt)

I use Avare (latest 10.0.5) on a new Samsung Tab A7 tablet.

Thanks,
Mark

mhrussell1, 2021-06-02:
While the last comment above is correct about the power LED.  That did not completely solve my problem. After using this for 4 different 4 hour banner tows in SoCal area the symptom seems to be that when over land, I lose the wifi connection to both my phone and tablet. When over the water at 500 feet, it works just fine. Seems like this could be more of a phone/tablet problem with wifi interference. It behaves the same on two different Stratux boxes with two different versions of 1.6r1 and different SDRs.  I just changed the location setting to en-US from en-GB suggested somewhere else.  I will test that this weekend on two more 4 hour flights.  Any suggestions for what to look for in the logs would be helpful.  I know close to exact times when the wifi turns off based on my GPS track.

mhrussell1, 2021-06-06:
Setting the country code in the file did not help.  Still had the wifi drop out over land at 1000 feet.